jrkerns/pylinac #14: bug in report_basic_parameters() ?

robmarkcole, 2015-04-16:
this function works fine with the demo file, but I get the error 
IndexError: arrays used as indices must be of integer (or boolean) type
when I call it on my own data file, at the link below
https://www.dropbox.com/s/mj77xdjqjx4f5qr/ZZ.bin?dl=0

jrkerns, 2015-04-16:
This was indeed a bug. In version 3 of trajectory logs there are a few changes vs. version 2. While I accounted for most of the changes, I didn't account for the extra couch axes (roll and pitch). These have been corrected for. Your log runs now. I'll check for other v2/v3 changes that I might've missed, but for now this is working fine. You can reinstall pylinac [from the repo](http://pylinac.readthedocs.org/en/latest/installation.html#i-know-python-already).

closed 2015-04-16 by jrkerns

robmarkcole, 2015-04-16:
Excellent thanks. I try to install from the repo and get this error, uncertain how to proceed, any advice appreciated. 
Cheers

  Cannot unpack file C:\Users\RCole02\AppData\Local\Temp\pip-zgssb6m4-unpack\pyl
inac.git (downloaded from C:\Users\RCole02\AppData\Local\Temp\pip-x6i3bx1l-build
, content-type: text/html; charset=utf-8); cannot detect archive format
  Cannot determine archive format of C:\Users\RCole02\AppData\Local\Temp\pip-x6i
3bx1l-build

jrkerns, 2015-04-16:
When I wrote the docs for installing via the github repo it worked the way stated, but I ran into the same error just now. [This answer](http://stackoverflow.com/a/8382819/3060280) helped and is apparently [faster](http://stackoverflow.com/a/24811490/3060280). Also, I ran into an error until I upgraded pip to the latest version (6.0.8>6.1.1). 

So, `pip install pip --upgrade` to upgrade pip, then `pip install https://github.com/jrkerns/pylinac/zipball/master`

Sorry for all the rigamarole.

commit fc3a12f7 (2015-04-16): closes #14.