jrkerns/pylinac #76: load images from BytesIO

spitzbubchen, 2016-09-20:
Is it possible to load images directly from BytesIO? For example, this
```
	temp = BytesIO()
	temp.write(data)
	temp.seek(0)
	star = Starshot(temp)
```
leads to the following error:

`TypeError: argument should be string, bytes or integer, not _io.BytesIO`

jrkerns, 2016-09-20:
Initially, I did make the major pylinac classes compatible with Bytes objects. It became untenable when I started trying to also deal with saving files, anonymizing, loading URLs, etc. It was much easier to always deal in out-of-stream files. I also couldn't find anyone who used it in the case you are describing. 

Could writing to temporary files work for you? Something like this [gist](https://gist.github.com/jrkerns/fcc1f011233e85aaf8da62bba373a6d7).

spitzbubchen, 2016-09-21:
Okay, I see your reasons for not doing this. As a work around I'm writing files to disk which works fine although the `star.save_analyzed_image` process is rather slow... It also looks like my solution is similar to your gist so I'll close this issue.
```
with open(os.path.join(dir, 'temp-starshot.jpg'), 'wb') as temp:
	temp.write(file.data.read())
star = Starshot(os.path.join(dir, 'temp-starshot.jpg'), dpi = 96, sid = 1000)
star.analyze()
print(star.return_results())
star.save_analyzed_image(os.path.join(dir, 'temp-starshot.png'))
```

closed 2016-09-21 by spitzbubchen