jrkerns/pylinac #87: additional star shot output

spitzbubchen, 2017-04-13:
In addition to the min circle that touches all lines would it be possible to add output for the:
- min circle that emcompasses all intersections points
- distance between the two furthest separated intersection points

jrkerns, 2017-04-16:
That would actually be a non-trivial amount of work. Can you tell me if this is common practice or what extra diagnosis you can get with that info?

spitzbubchen, 2017-04-24:
I would say common practice. RIT for example provides these outputs.

jrkerns, 2017-04-25:
I'm still confused. If, e.g., I take 3 images at the collimator cardinal angles and they have 1mm of offset from the nominal isocenter then my standard circle will have a radius of 1mm. If the circle encompasses the intersections then it'll have a radius of sqrt(2). However, if I take 6 images which includes 45 degree images and they also have an offset of 1mm then my standard circle would (correctly) still have a radius of 1mm, but my circle that encompasses all intersections would now indicate sqrt(1.25). So such a value depends on the number of images I take. This could be exaggerated even more: assuming that same 1mm offset, take images at 0, 89, and 269. The 89/269 intersection would be way far away from the CAX, but it doesn't mean that they are really off by that much. 

What clinical decisions have been made from this information? If my "max intersection" circle has a large value how am I to remedy that? What mechanical component is failing if my intersection circle is high but my standard circle is well within range?

spitzbubchen, 2017-05-14:
I agree with your argument that star shot analysis should not be angle dependent as this reduces potential errors in interpretation and allows better comparison of data between machines. Pipspro also uses the minimum circle touching all lines method. Closing the issue.

closed 2017-05-14 by spitzbubchen