jrkerns/pylinac #92: Gui terminating with uncaught exception of type NSException

robmarkcole, 2017-04-28:
OSX Sierra, pylinac 2.0.1

```
>>> import pylinac
>>> pylinac.gui()
2017-04-28 15:21:21.571 python3[1129:94855] -[NSApplication _setup:]: unrecognized selector sent to instance 0x104a531d0
2017-04-28 15:21:21.574 python3[1129:94855] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplication _setup:]: unrecognized selector sent to instance 0x104a531d0'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff8027537b __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fff9506948d objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff802f6f04 -[NSObject(NSObject) doesNotRecognizeSelector:] + 132
	3   CoreFoundation                      0x00007fff801e7805 ___forwarding___ + 1061
	4   CoreFoundation                      0x00007fff801e7358 _CF_forwarding_prep_0 + 120
	5   Tk                                  0x000000010af14c02 TkpInit + 471
	6   Tk                                  0x000000010ae902a9 Tk_Init + 1794
	7   _tkinter.so                         0x000000010a8e3144 Tcl_AppInit + 84
	8   _tkinter.so                         0x000000010a8e2e39 _tkinter_create + 1129
	9   libpython3.5m.dylib                 0x000000010004ff38 PyCFunction_Call + 280
	10  libpython3.5m.dylib                 0x00000001000bd2df PyEval_EvalFrameEx + 22431
	11  libpython3.5m.dylib                 0x00000001000c10c3 _PyEval_EvalCodeWithName + 1779
	12  libpython3.5m.dylib                 0x00000001000b7b1e PyEval_EvalCodeEx + 78
	13  libpython3.5m.dylib                 0x000000010003430f function_call + 351
	14  libpython3.5m.dylib                 0x000000010000fd73 PyObject_Call + 99
	15  libpython3.5m.dylib                 0x000000010002130d method_call + 141
	16  libpython3.5m.dylib                 0x000000010000fd73 PyObject_Call + 99
	17  libpython3.5m.dylib                 0x0000000100065add slot_tp_init + 125
	18  libpython3.5m.dylib                 0x0000000100062329 type_call + 281
	19  libpython3.5m.dylib                 0x000000010000fd73 PyObject_Call + 99
	20  libpython3.5m.dylib                 0x00000001000bd766 PyEval_EvalFrameEx + 23590
	21  libpython3.5m.dylib                 0x00000001000c192f fast_function + 207
	22  libpython3.5m.dylib                 0x00000001000bd434 PyEval_EvalFrameEx + 22772
	23  libpython3.5m.dylib                 0x00000001000c10c3 _PyEval_EvalCodeWithName + 1779
	24  libpython3.5m.dylib                 0x00000001000b7ac1 PyEval_EvalCode + 81
	25  libpython3.5m.dylib                 0x00000001000e642c PyRun_InteractiveOneObject + 588
	26  libpython3.5m.dylib                 0x00000001000e5d5e PyRun_InteractiveLoopFlags + 206
	27  libpython3.5m.dylib                 0x00000001000e5c5c PyRun_AnyFileExFlags + 60
	28  libpython3.5m.dylib                 0x00000001000fcd47 Py_Main + 3591
	29  python3                             0x0000000100000dc7 main + 215
	30  python3                             0x0000000100000ce4 start + 52
	31  ???                                 0x0000000000000001 0x0 + 1
)
libc++abi.dylib: terminating with uncaught exception of type NSException
```

jrkerns, 2017-04-28:
was pylinac installed via conda or pip?

robmarkcole, 2017-04-28:
pip

jrkerns, 2017-04-28:
Hmm. That's a pretty interesting stacktrace lol. Never seen one quite like that. I can't reproduce but I also don't have OSX. I'd try either installing via conda to make sure the numpy/scipy/scikit's install/upgrade properly and/or make a new env and try it that way. Either way is to make sure nothing happened during the upgrade.

phillipdhwall, 2017-11-06:
Getting a similar error as well. OSX Sierra, pylinac 2.0.1. Installed with conda

[pylinac_gui_error.txt](https://github.com/jrkerns/pylinac/files/1447368/pylinac_gui_error.txt)

jrkerns, 2017-11-07:
Possibly this [bug](https://stackoverflow.com/questions/30031063/nsexception-with-tkinter-on-mac)

phillipdhwall, 2017-11-07:
So I've been messing around with this a bit. I updated to the latest ActiveTcl on my machine as, which didn't fix anything (also tried reinstalling python as well). After some scavenging I *think* I found the source of the problem. From this [matplotlib doc](https://matplotlib.org/tutorials/introductory/usage.html#what-is-a-backend) on backends:

> To make things a little more customizable for graphical user interfaces, matplotlib separates the concept of the renderer (the thing that actually does the drawing) from the canvas (the place where the drawing goes). The canonical renderer for user interfaces is Agg which uses the Anti-Grain Geometry C++ library to make a raster (pixel) image of the figure. All of the user interfaces except macosx can be used with agg rendering, e.g., WXAgg, GTKAgg, QT4Agg, QT5Agg, TkAgg.

With this and a little more messing around, I ended up modifying the init file to overwrite the matplotlib backend to 'TkAgg' and it works perfectly now. I'd be interested to see if this fixes the problem on all OSXs and doesn't introduce others in other operating systems. I'm a bit of a novice, but I've considered submitting a PR to solve this issue. But it's a very small fix and again, I'm not sure if it extends generally. Thoughts? Any other OSX users coming across this problem?

jrkerns, 2017-11-08:
Does ``webbrowser.open()`` work? I read thats a workable hack.

jrkerns, 2017-11-08:
That's interesting. Sounds like an easy fix if that's the source of the issue

Cheng-Tao, 2018-02-24:
You can fix this by creating a file named "matplotlibrc" in "/home/.matplotlib" folder and writing "backend: TkAgg" in it.