jrkerns/pylinac #93: Error in dynalog demo, 2.0.1

robmarkcole, 2017-04-28:
```
from pylinac import Dynalog
Dynalog.run_demo()

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-b0c74e74bcc2> in <module>()
      1 from pylinac import Dynalog
----> 2 Dynalog.run_demo()

/Users/robincole/anaconda3/lib/python3.5/site-packages/pylinac/log_analyzer.py in run_demo()
   1764         """Run the Dynalog demo."""
   1765         dlog = Dynalog.from_demo()
-> 1766         dlog.report_basic_parameters()
   1767         dlog.plot_summary()
   1768 

/Users/robincole/anaconda3/lib/python3.5/site-packages/pylinac/log_analyzer.py in report_basic_parameters(self, printout)
   1465             p95 = "95th percentile error: {:3.3f} cm\n".format(self.axis_data.mlc.get_error_percentile(95, only_moving_leaves=False))
   1466             num_holdoffs = "Number of beam holdoffs: {:1.0f}\n".format(self.num_beamholds)
-> 1467             self.fluence.gamma.calc_map()
   1468             gamma_pass = "Gamma pass %: {:2.2f}\n".format(self.fluence.gamma.pass_prcnt)
   1469             gamma_avg = "Gamma average: {:2.3f}\n".format(self.fluence.gamma.avg_gamma)

/Users/robincole/anaconda3/lib/python3.5/site-packages/pylinac/log_analyzer.py in calc_map(self, doseTA, distTA, threshold, resolution, calc_individual_maps)
    634         actual_img = image.load(self._actual_fluence.array, dpi=25.4 / resolution)
    635         expected_img = image.load(self._expected_fluence.array, dpi=25.4 / resolution)
--> 636         gamma_map = actual_img.gamma(expected_img, doseTA=doseTA, distTA=distTA, threshold=threshold)
    637 
    638         # calculate standard metrics

/Users/robincole/anaconda3/lib/python3.5/site-packages/pylinac/core/decorators.py in wrapper(*args, **kwargs)
     73                         if value not in bound_values[name]:
     74                             raise ValueError("Argument '{0}' must be one of {1}".format(name, bound_values[name]))
---> 75             return func(*args, **kwargs)
     76         return wrapper
     77     return decorate

/Users/robincole/anaconda3/lib/python3.5/site-packages/pylinac/core/image.py in gamma(self, comparison_image, doseTA, distTA, threshold)
    574         # set up reference and comparison images
    575         ref_img = ArrayImage(copy.copy(self.array))
--> 576         ref_img.check_inversion()
    577         ref_img.ground()
    578         ref_img.normalize()

/Users/robincole/anaconda3/lib/python3.5/site-packages/pylinac/core/decorators.py in wrapper(*args, **kwargs)
     43                             if value not in bound_types[name]:
     44                                 raise TypeError("Argument '{0}' must be {1}".format(name, bound_types[name]))
---> 45             return func(*args, **kwargs)
     46         return wrapper
     47     return decorate

/Users/robincole/anaconda3/lib/python3.5/site-packages/pylinac/core/image.py in check_inversion(self, box_size, position)
    528         lt_lower = self.array[-row_pos-box_size:-row_pos, col_pos: col_pos+box_size]
    529         rt_lower = self.array[-row_pos-box_size:-row_pos, -col_pos-box_size:-col_pos]
--> 530         avg = np.mean((lt_upper, lt_lower, rt_upper, rt_lower))
    531         if avg > np.mean(self.array.flatten()):
    532             self.invert()

/Users/robincole/anaconda3/lib/python3.5/site-packages/numpy/core/fromnumeric.py in mean(a, axis, dtype, out, keepdims)
   2887 
   2888     return _methods._mean(a, axis=axis, dtype=dtype,
-> 2889                           out=out, **kwargs)
   2890 
   2891 

/Users/robincole/anaconda3/lib/python3.5/site-packages/numpy/core/_methods.py in _mean(a, axis, dtype, out, keepdims)
     76     elif hasattr(ret, 'dtype'):
     77         if is_float16_result:
---> 78             ret = a.dtype.type(ret / rcount)
     79         else:
     80             ret = ret.dtype.type(ret / rcount)

AttributeError: 'tuple' object has no attribute 'dtype'
```

jrkerns, 2017-04-28:
could be because i bumped the numpy version. what version do you have?

robmarkcole, 2017-04-28:
Hmm no sure why I have 2..

[image]

closed 2020-11-29 by jrkerns