jrkerns/pylinac #101: Winston Lutz Accuracy

Quixeh, 2017-08-09:
Hi All, 

I've encountered a situation where the Winston-Lutz optimization didn't act as expected, and on further investigation, I wanted to discuss with the wider community. 

I gave the algorithm four images at cardinal gantry angles (0, 90, 180, 270) from an old linac with a radiation isocentre diameter in the 1-2mm range. The G0 and G180 images gave CAX positions which differed relative to the BB by ~1.2mm, and the lateral images differed by ~0.5mm. 

The algorithm works by minimizing the maximum distance from the radiation isocentre point to any line it finds. Given four images like that, once the radiation isocentre point is located between the two projections with the larger distance between them, there is nothing to push the optimization to minimize the distance to the projections with smaller distances between them - in my case the algorithm essentially ignored these projections. See the image below - the red dot is the calculated radiation isocentre, the blue cross is the BB, the red and orange lines are the AP projections and the green and blue lines are the lateral projections. 

[image]

Would it be more appropriate if the minimization was looking to reduce the average distance?

jrkerns, 2017-08-15:
I'm not sure what your clinic's practice is, but our is worried about the maximum distance away from the BB.  If the test is done by iteratively moving the BB, the maximum distance will converge to a location in between the two most distal points. 

If you are interested in the average that should be a relatively simple change. Can you provide a paper or presentation that argues for the average?

Quixeh, 2017-08-15:
Hi James, 

We'd also use the same technique, but the BB position isn't important here - the minimisation would never force the position of the reported Radiation Isocentre to be correct in 'HGT' on my image, unless the BB just happened to be at the exact HGT. 

Du et al. (A simple method to quantify the coincidence between portal image
graticules and radiation field centers or radiation isocenter) MedPhys 37 2010 - They're using the method we're interested in, but instead of just looking at the maximum or the average they define a cost function which is a combination of the average, maximum and stdev. Perhaps that is the direction we should move in?

jrkerns, 2017-08-15:
Yeah I tried that one time and the results were effectively the same. If I get time this weekend I can play around with the results of the two methods.

commit da455feb (2018-05-11): Closes #124. Closes #116. Closes #101.
closed 2018-05-11 by jrkerns