jrkerns/pylinac #102: Planar Imaging module  LeedsTOR.publish_pdf bug

danysimard, 2017-08-31:
Hello, 
I thing there is an issue with LeedsTOR.publish_pdf.    When I run your demo, I get an error at line 911.  I guess it's a copy and paste error from StandardImagingQC3 publish_pdf.

Great work by the way.

Dany

Python 3.4.5 (default, Jul 15 2016, 16:39:07) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pylinac.flatsym import BeamImage
>>> from pylinac import LeedsTOR
>>> leeds=LeedsTOR.from_demo_image()
>>> leeds.analyze()
>>> leeds.publish_pdf('leeds_october16.pdf')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/dsimard/pylinac/env/lib/python3.4/site-packages/pylinac/planar_imaging.py", line 911, in publish_pdf
    'MTF 80% (lp/mm): {:2.2f}'.format(self._mtf(80)),
AttributeError: 'LeedsTOR' object has no attribute '_mtf'

danysimard, 2017-08-31:
Ok, fixed it by adding the copy _mtf from StandardImagingQC3 to LeedsTOR

    def _mtf(self, x=50):
        norm = max(roi.mtf for roi in self.hc_rois)
        ys = [roi.mtf / norm for roi in self.hc_rois]
        xs = np.arange(len(ys))
        f = interp1d(ys, xs)
        try:
            mtf = f(x / 100)
        except ValueError:
            mtf = min(ys)
        return float(mtf)

danysimard, 2017-08-31:
By the way, I also changed the precision for contrast in the report
'Median Contrast: {:2.3f}'

jrkerns, 2018-03-01:
Yes, thanks for this catch. Closed in #94

closed 2018-03-01 by jrkerns