jrkerns/pylinac #111: TG51 class PDD and TPR logic limits do not match documentation

jmartens, 2017-11-10:
The PDD and TPR limits specified in the documentation of the code do not match the program logic as specified in the code:

Inline documentation states:

```
pddx : {>=0.627, <=0.861}
    The *PHOTON-ONLY* PDD measurement at 10cm depth for a 10x10cm2 field.
tpr : {>=0.623, <=0.805}
    The TPR ratio of the 20cm measurement divided by the 10cm measurement.
```

The code logic implements the following correct values as limits:

```
PDD_LOW = 62.7
PDD_HIGH = 86.1
TPR_LOW = 0.623
TPR_HIGH = 0.805
```
But then seems to disregard the limits as the comparison does not include them, since it uses greater than and less than operators instead of greater or equal and less than or equal:
`if pddx > PDD_HIGH or pddx < PDD_LOW:` and `if tpr > TPR_HIGH or tpr < TPR_LOW:`

jmartens, 2017-11-10:
I am not sure what TG51 report states as I do not have a copy at hand ATM, but I guess you can check and apply the proper fix, either in code logic or in the inline docs :-)

jrkerns, 2017-11-10:
The TG-51 addendum says it's valid for PDDs between 63% to 86% (Table I). The paper that those values are based on, Muir & Rogers 2010, state that the same values are valid for PDDs between 62.7% and 86.1% (Table III) and is what pylinac uses. Perhaps the TG addendum was being conservative by rounding, and I suppose that's what should happen here. In practice, these limits are rarely hit. I will update the docs and code to follow the addendum values, which seem to be inclusive ("...are valid for 63 <%dd(10)X < 86..."). Thanks for the detailed catch!

commit 93632d0d (2017-11-10): Documentation updates; PDD kQ limits adjusted to TG-51 addendum values. Closes #111.
closed 2017-11-10 by jrkerns