jrkerns/pylinac #113: Python 3.6

mattcschmidt, 2017-11-21:
Hello,
I'm currently showing off pylinac to dozens of people every month as a tool to read and analyze trajectory log files and assist in the analysis of Winston Lutz tests. Currently, in my workflow, the most time consuming part is downloading anaconda, and then installing a new python environment for 3.5 to get the libraries to work. Is there any case we can get pylinac working in 3.6 or maybe I am doing something wrong? Thanks!
[image]

jrkerns, 2017-11-21:
Of course! Pylinac itself is pure python so it can be installed via pip. The reason conda is nice is because it will also take care of the numpy, scipy, and scikit-x dependencies. It sounds like you want to avoid making a separate environment and just start running from the base environment, so I will give the commands for that, otherwise I'd recommend creating a new environment:

`# after fresh miniconda python 3.6 install`
`conda install numpy scipy scikit-learn scikit-image`
`pip install pylinac  # will install all other dependencies`
`# good to go!`
`python`
`...`

As an aside, I am dropping MPLD3 as a dependency in the next version of pylinac as well as enhancing the WL module to be a bit simpler and able to be used on a day-of interactive BB shift session.

mattcschmidt, 2017-11-22:
This is awesome and super easy. Thanks a ton. Do you mind if I share with your privately some of the documents I've made that I follow when showing this to students and clinical physicists. I'd love to have your input or any additions that you think would be helpful for the general public. We've recently started teaching some SRS and IMRT courses in China, and they seem to think pylinac is extremely helpful. My email is mattcschmidt@hotmail.com if that is something you'd be willing to do.

jrkerns, 2017-11-22:
Please do!

mattcschmidt, 2018-01-13:
Hi James,

Its me again. I have another question coming from some of our Chinese physics friends regarding verification of small fields for SRS planning. I was wondering if there were any workflows that you had seen in Pylinac to perform the verification of dose and position for small fields (specifically those delimited by conical collimators) using film.
I noticed you had some film analysis for symmetry and flatness, but didn't want to spend too much time trying to adapt that if you knew of a better workflow off-hand.

jrkerns, 2018-03-01:
Sorry for the late response. What type of verification are you referring to or trying to do? like a max pixel value of a film shot with a certain size cone? Is it different than a WL test?

mattcschmidt, 2018-03-01:
Hi James,
Maybe slightly different. I think I should have said dose and shape instead of dose and position. Of course WL can give the positional accuracy, but they were looking for something that can help them compare the dose output and fluence shape of small fields. I have been showing the classes I teach in China some of your open source projects for quite some time now, and they've been received fantastically. Thanks!

jrkerns, 2018-03-01:
You could fairly easily extract profiles from a digitized film if the flatness/symmetry module was not appropriate. A skeleton would look like:
-using [image.load()](https://pylinac.readthedocs.io/en/latest/core_modules.html#pylinac.core.image.load) to load the digitized film
-using [SingleProfile()](https://pylinac.readthedocs.io/en/latest/core_modules.html#pylinac.core.profile.SingleProfile) at a known row or column (you could make one for each) position and
-compare to a known profile (perhaps a reference film and then pickle the result or dump to CSV/Excel)

mattcschmidt, 2018-03-01:
Great. Let me pass this along. I've never used any of the film analysis stuff (not using film).
Thanks!

closed 2018-05-10 by jrkerns