jrkerns/pylinac #116: Winston Lutz : underestimated bb_shift_vector value

midamo, 2018-02-16:
Hi James, 

I really appreciate the new features that you introduced since our last discussions. I am discovering and testing the new features as part of a new QA procedure that I am working on. 

The bb_shift_vector function is very usefull, but it gives slightly inaccurate results.   
```
  def bb_shift_vector(self):
        """The shift necessary to place the BB at the radiation isocenter"""
        tv = Vector()
        vs = [img.cax2bb_vector3d for img in self.images]
        for v in vs:
            tv += v
        return Vector(-tv.x / len(vs), -tv.y / len(vs), -tv.z / len(vs))

```
Now let's assume that you have a bb positioned 1mm UP from the "real isocenter" in a 4 gantry angle Winston Lutz (0, 90,180 and 270). The cax2bb_vector3d would indeed be 1mm for 2 of the 4 gantry angles, but would be 0 mm for the images taken at gantry = 0 and 180 degrees. In consequence, the  bb_shift_vector would be 0.5 mm instead of the expected 1mm. 

The problem is simply that cax2bb_vector3d returns "0" when it simply cannot evaluates the position in a specific axis. This function could output a "NAN"... or you could modulate a summation weight on "tv" based on the sine/cosine of the gantry angle. 

Do you come to the same conclusions ? What is your opinion ?

jrkerns, 2018-03-01:
I do have to agree with you, although in my comparisons to other commercial software and personal use it has worked out quite fine, so I will have to run some experiments with the BB shifted in various axes. I have several but these involve several axes at once.

jrkerns, 2018-03-31:
Confirmed it's an issue. Working on a fix.

commit da455feb (2018-05-11): Closes #124. Closes #116. Closes #101.
closed 2018-05-11 by jrkerns