jrkerns/pylinac #132: Refactor PDF module

jrkerns, 2018-06-16:


keithoffer, 2018-09-04:
I was looking at doing some work to improve the PDF reports (in particular fix things like long file names squishing the images), but noticed that you've set this issue up. What do you plan on refactoring? Are you planning to modify the file reports being generated as part of the refactoring process?

jrkerns, 2018-10-24:
Hi Keith, sorry for the super late reply. 

I needed to refactor it to modularize the functions and so forth. Each module had all kinds of spaghetti code using pdf functions. The long file names is indeed an issue I've seen. I was thinking about possibly just using the last ~30 characters or so of the full file name. That way it wouldn't overflow.

commit 78b4dbf2 (2018-10-24): Closes #132
closed 2018-10-24 by jrkerns