jrkerns/pylinac #135: Potential to use ScriptedForms to create online demos?

SimonBiggs, 2018-08-28:
Hi @jrkerns,

I'm not sure if you would be keen, but I have been using PyLinac to create a demos of using ScriptedForms. For example:

https://mybinder.org/v2/gh/SimonBiggs/scriptedforms-examples/master?urlpath=scriptedforms/use/pylinac.form.md

If you're interested I would be happy to make some more if you wanted to use them for pylinac docs.

Cheers,
Simon

jrkerns, 2018-08-28:
Really cool! I've been meaning to explore the scriptedforms repo; clearly it's gaining traction. I think something like this would be great for a small clinic that wants to have some type of GUI for pylinac that's not 80's tkinter style =)

SimonBiggs, 2018-08-28:
Thanks James :).

Hopefully also the underlying code that updates/changes the GUI is more
easily parsed and edited by non-programmers. That's the plan anyway.

Here's the file that creates that form:
https://raw.githubusercontent.com/SimonBiggs/scriptedforms-examples/master/pylinac.form.md

closed 2020-06-01 by SimonBiggs