jrkerns/pylinac #148: Add Support for PTW EPID QC Phantom

tbezo, 2018-10-31:
Hi.

In our clinic we are using equippment from PTW.
Would it be a lot of work to add support for the PTW EPID QC Phantom (http://www.ptw.de/2857.html)?

I uploaded a DICOM Image from that phantom to your Dropbox if you would like to have a closer look.

jrkerns, 2018-10-31:
Thanks for that. Although I have a big backlog I'm always willing to add new features. 

However, I would need a range of images to evaluate so that the algorithm is robust and can handle varying circumstances. If you can take images at a few different SSDs (assuming phantom at iso), a few with rotations relative to the panel, and a few with different acquisition techniques (kVp, mAs, etc) that would give me a range of images to test against.

tbezo, 2018-11-08:
Hi.
Thanks for your quick response. I am very thankful that you are considering to add the PTW Phantom.  Just put it at the End of your list or wherever you deem fit.

Hopefully I will be able to upload more images soon. 
The EPID QC Phantom is only specified for 4-25 MV beams, so I would skip the different mAs/kVp values. And because it is quite large (25 cm x 25 cm) I will have to use our Truebeam with it's larger EPID to create images with different Source Imager Distances.

randlet, 2021-04-23:
@tbezo  I didn't realize there was a feature request open for this.  A while ago I added some experimental support for the PTW EPID QC here: https://github.com/qatrackplus/pylinac/blob/qat_2.5.0/pylinac/planar_imaging.py#L952 but it is not well tested at this point.   Perhaps you could try it out at some point.

MohammadSalehi72, 2021-05-17:
> @tbezo I didn't realize there was a feature request open for this. A while ago I added some experimental support for the PTW EPID QC here: [...]

Dear Randlet;

Thank you for the link. Did you test the "PTWEPIDQC" class for the real application?  and may I asked you to share with me some examples of PTW EPID QC Phantom images?

randlet, 2021-05-19:
Hi @MohammadSalehi72,  the analysis is still being evaluated, but I do not have images I can share.