jrkerns/pylinac #150: Make it possible to pass file like objects (in addition to filenames) to anaylysis classes

randlet, 2018-11-02:
**Is your feature request related to a problem? Please describe.**

Sometimes you have a file object in memory that doesn't exist on disk which means you can't use e.g. the picket fence module.

**Describe the solution you'd like**

It's common (see e.g. [pydicom.read_file](https://github.com/pydicom/pydicom/blob/master/pydicom/filereader.py#L778)) to accept either a file name or a file-like object.  So then you can do e.g.

```
foo = io.BytesIO(...)
picket = pylinac.PicketFence(foo)
```

**Describe alternatives you've considered**

You can write the file to disk beforehand but that's not efficient or always practical.

jrkerns, 2019-01-25:
So, the good news here is that pylinac uses pydicom for dicom images, which are the majority of images used. I need to think about a few things though as there are certain properties that are based on the file name and base folder (e.g. where to save the PDF report if it's not passed) but this should be doable. The biggest change is in how the watcher process handles this, but it needs some help anyway.

This would be difficult with dynalogs as they come in pairs and the user would have to pass in both logs at once as the current behavior is to look for the other half when one half is passed in.

randlet, 2019-01-25:
Thanks James.  Sounds like the change would be a bit more involved for dynalogs which I hadn't thought of.  With regards to saving the PDF report, not sure if it's feasible in this case but you could write the PDF to an io.BytesIO() object and return that to the user rather than writing to disk.

FWIW, this request was motivated by wanting to use AWS S3 for handling file uploads with QATrack+ when there are multiple app servers handling requests (since you can't always count on a request going to the same server where the original file was uploaded).  I ended up realizing there were some other similar issues with QATrack+ and eventually switched to AWS EFS which made this problem go away.  Still, a nice to have feature where possible!

jrkerns, 2019-08-22:
Closing until another use case pops up.

jrkerns, 2021-05-18:
A case popped up.

commit f902b08d (2021-05-18): Add support for file objects. Closes #150.
closed 2021-05-18 by jrkerns