jrkerns/pylinac #157: Dynalog MLC Leaf Position Differences

bvangenderen, 2018-11-30:
For MLC positions extracted from a Dynalog file, the differences appear to be calculated using the expected value and the actual value from the same row (measurement) of the Dynalog. However, based on Dynalog documentation, the expected value is referring to the value the machine expects the leaf to have at the **next** measurement (see p.33 of the Dynalog Viewer Reference Guide if available).

This error should only affect logs loaded from Dynalogs, I believe they have lined up the data more sensibly in Trajectory files but I haven't had the opportunity to investigate those yet.

Possible fix is to shift the expected leaf data down by one to realign it with its corresponding actual value. As a placeholder solution on my end, I'm currently duplicating the first position and removing the last position from each leaf's expected array and then calculating the differences as normal (actual - expected).

jrkerns, 2018-12-08:
Hi Brandon, 
    I talked to my coworker at Varian about this as I wanted to see what another person thought. We both read pg 33 and we both came to the same but opposite conclusion as you. I've been racking my brain working through their very convoluted example. I can see your point of view but I believe the behavior to be like this:
-At a given snapshot point (I guess what they call transition point), there is an expected MLC position and actual MLC position record. For the sake of simplicity let's say they are both 0.0mm for that snapshot.
-Immediately after the snapshot recording, when the MU has reached the dose fraction it's been waiting for, the new MLC position is given to the controller to order the MLCs ("If so, the next dose fraction, along with a new set of leaf positions, is read from the MLC plan file"). Let's say the next expected MLC position is 0.1mm.
-From the time immediately following the transition point the expected position of the MLCs is always known by the controller. From the moment following the transition point to the next snapshot, the controller will try to drive the MLCs to the always-known expected position. 
-After 50ms, the expected position (which was always known to the MLCs for those 50ms which follows from the above quote) is recorded in the next snapshot along with the actual position of the MLCs at that moment in time.

Thus, we believe the actual/expected difference at a given snapshot is the true deviation. I.e. we do not believe the controller is given a "grace" snapshot to get into position.

bvangenderen, 2018-12-10:
Hi James,
Thanks for looking into it further. I agree that the example on page 33 of the guide is rather convoluted, but I'm still not quite convinced, for two reasons.

I was initially pointed to this different method of calculating the leaf  position errors when I was reading the manual for PipsPro v5.5 (Standard Imaging) which has a logfile analysis module. This isn't directly from Varian of course, but It has appendices detailing their calculations and on p.  76 it mentions: 

> In both DynaLog and Trajectory files, errors in MLC leaf positioning
(and speed) are determined by subtracting expected positions from
actual positions. Equation 1 shows how errors for an individual leaf
are calculated using DynaLog files. It shows that the error for any leaf
m at time n is calculated using the current actual position (at time n)
and the expected position for the previous sample (at time n-1).

> _Error(m,n) = LeafPositionActual(m,n) - LeafPositionExpected(m,n-1)_

> Equation 2 shows how errors for an individual leaf are calculated using
Trajectory Logfiles. It shows that the error for any leaf m at time n is
calculated using the current actual position (at time n) and the expected
position for the same sample (also at time n).

> _Error(m,n) = LeafPositionActual(m,n) - LeafPositionExpected(m,n)_

This is 3rd party documentation though, so I opened a Varian support ticket to try to get clarification from them directly and this was their response:

> •	 In the Trilogy Dynalog files, the plan (order) values and the actual (status) values do not line up the same way the TrueBeam Trajectory logs line up.
•	In the TrueBeam file, each line of data is a consistent set. The ordered value and the status value are taken at the same instance.
•	In the Trilogy file, each line of data is NOT a consistent set.
•	The plan (ordered) value from the previous line needs to be analyzed with the current line’s actual (status) value to give similar results.
•	This is because the plan position logged in a data line is where we want leaves to be in 50 msec, not where we asked them to be now.
•	Also see p. 33 of Dynalog Viewer Ref Guide 
   o	In each 50-ms control cycle, the Clinac reports the current dose fraction along with
the actual position of each leaf. 
        The MLC controller determines if the actual dose
fraction has reached the next dose fraction listed in the MLC plan file. 
        If so, the next dose fraction, along with a new set of leaf positions, is read from the MLC plan file; these positions then become the new expected leaf positions. 
    o	At this transition point, the leaves are in the leaf positions associated with the former dose fraction the old positions but the MLC controller now expects them to be at the new positions.
    o	Often for the step-and-shoot technique, these position changes are large on the order of millimeters or even centimeters —and they exceed the dynamic leaf tolerance. In this
case, a beam hold-off is asserted. 
        At this instant in time, the following occurs:
• Leaf position deviations are equal to the distance between the current actual leaf positions (which were the old expected positions) and the newly calculated expected leaf positions. 

We use Clinac iX machines at our centre but I think the Dynalog format is the same for both them and the Trilogy machines. With your contacts at Varian, is there a way we can verify whether the info I received in the support ticket is accurate?

jrkerns, 2018-12-10:
Hmmm. Very interesting. Let me ask around!

jrkerns, 2018-12-14:
It would appear that you are right and that the expected value in the dynalog is not actually the expected value. Will fix.

closed 2019-01-25 by jrkerns

jrkerns, 2019-03-21:
Until proven otherwise this issue will be reverted. The dynalog file viewer and other software give error metrics that match the original behavior. At the very least these softwares should match. If they are all wrong then at least pylinac is not the only one.

bvangenderen, 2019-03-21:
Noted. I'll keep looking into the discrepancies in error calculations and let you know if I find anything.

commit ee86c4d9 (2019-01-25): Closes #157