jrkerns/pylinac #158: Catphan analysis : fantom roll

jlley, 2019-01-23:
**Describe the bug**
A clear and concise description of what the bug is.

The function find_catphan_roll() in ct.py doesn't work properly everytime. When we analyze the Catphan 504 images, the roll angle is sometimes around 13 degrees even if the fantom is well positionned. 
We found that it come from the detection of the air insert at 12h. If the value from the is_right_area() is out of the range for the threshold, the program takes the closest insert instead of the air insert.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

In the function : def find_phantom_roll(self): and then def is_right_area(region):
Could you change the inferior limit from 1.5 to 2 please ?
    original code:   return thresh * 1.5 > region.filled_area > thresh / 1.5 
    update:  return thresh * 2 > region.filled_area > thresh / 1.5 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jrkerns, 2019-01-25:
Looks like this would work. Need to run it on my larger private dataset though. #159

jrkerns, 2019-02-25:
Will be in v2.2.3

closed 2019-02-25 by jrkerns