jrkerns/pylinac #160: Dynalog Bank B Leaf Order

bvangenderen, 2019-02-15:
Possible discrepancy between the Log Analyzer documentation and the implementation for indexing leaves when loading Dynalog files. When reading the documentation:
[image]

I was expecting to get the following leaf pairs (for a standard 120 leaf MLC):
[image]

But the ordering I seem to be getting is like this:
[image]

I think this is the responsible code:
[image]

jrkerns, 2019-02-23:
Why were you expecting the first ordering?

mchamberland, 2019-02-23:
The documentation explicitly says that leaves 1 and 120, 2 and 119, and 60 and 61 pairs, which corresponds to the first ordering. Also, the documentation says (A_leaf, B_leaf) = (n, N_leaves +1 - 1), which agrees with the first ordering...

jrkerns, 2019-02-23:
Ah, I must be dyslexic. Got the first and second ordering reversed. Maybe I should read twice first >.>

jrkerns, 2019-02-23:
Issues should be fixed, looking at one more thing to double check. Resulting fluence should be the same as before if I've not missed anything. Thanks!

jrkerns, 2019-03-11:
So after further investigation, I realized that both Dynalogs and Trajectory logs worked the way you pointed out @bvangenderen, where documentation said pairs were 1/120, 2/119, ... . After I changed the behavior of dynalogs to match the behavior listed in the documentation I only now realized that the trajectory log behavior was affected. Rather than change both dynalogs and trajectory logs to match the doc behavior, which would be more work for both classes, I will revert the code change and change the documentation to be of the 1/61, 2/62 type. I will be adding more rigorous tests to the fluence calculations to detect these changes. Until now I was spot checking one value for ease. 

TL;DR It's easier to alter the documentation to the current/past behavior than vic versa.

commit c6f60220 (2019-02-23): Fixes leaf offset mismatch from documentation. Closes #160
commit 01cfbf02 (2019-03-11): Closes #160
closed 2019-03-11 by jrkerns