jrkerns/pylinac #161: Add support for Trajectory Log v4.0

jrkerns, 2019-02-23:


mattshrop, 2020-07-23:
Hi, do you still need some logs? I have some available. Is there PID in the logs? If I convert them to CSV I can't see any patient data but wan't to be sure before sending.
Regards

crcrewso, 2020-07-23:
> Hi, do you still need some logs? I have some available. Is there PID in the logs? If I convert them to CSV I can't see any patient data but wan't to be sure before sending.
> Regards

Matt,

If you're worried about PID, which I would be, you should be able grab trajectory logs from QA plans like DMLC or DRGS in the DailyQA folder inside <machine #>/Trajectory logs, assuming you run tests like that at least monthly. I'd send mine but we're still on 2.5, for now

mattshrop, 2020-07-23:
> Matt,
>
> If you're worried about PID, which I would be, you should be able grab trajectory logs from QA plans like DMLC or DRGS in the DailyQA folder inside <machine #>/Trajectory logs, assuming you [...]

Thanks, I might have to run a QA beam specifically for this - we use file mode for all our QA as it is so much quicker to load but file mode does not generate any logs. All of the logs saved are patient ones.

keithoffer, 2021-06-04:
I noticed the metadata fields weren't present today while messing around with some V4.0 trajectory logs. Do you still need some of these logs or do you have access to them? Unfortunately I can't contribute code due to an IP agreement in my current contract, but I could organise some V4.0 trajectory logs if you still need them.

jrkerns, 2021-06-04:
That would be great. I think I may have some but it's always better to have
more to validate on. You can upload them here:
https://www.dropbox.com/request/YKRu4AmuPsXu55uQq761 Thanks.

crcrewso, 2021-06-07:
> That would be great. I think I may have some but it's always better to have more to validate on. You can upload them here: https://www.dropbox.com/request/YKRu4AmuPsXu55uQq761 Thanks.
> […](#)

I just uploaded a few logs from our two TrueBeams, these were from an in house plan on an in house phantom recorded in MachineQA file mode as part of our Daily QA procedure, and are entirely kV/MV imaging based. 

Hopefully these are of some use.

jrkerns, 2021-06-08:
Good news, I was able to adjust the analyzer to handle v4.0 logs. The change is nearly trivial but took me longer than expected because the Varian log file spec is incorrect about the section sizes in v4.0 which took me some time to hunt down. V4.0 log handling should be in the v3.0 release. I will be releasing a release candidate for it soon as it has some big changes.

commit 9f13951b (2021-06-09): Closes #161
commit c908c420 (2021-06-09): Merge pull request #395 from jrkerns/feature/161_v4.0_logs

Closes #161
closed 2021-06-09 by jrkerns