jrkerns/pylinac #163: Asymetric jaw matching module.

peterg1t, 2019-02-25:
Hi James,
Here at the Tom Baker Cancer Center in Alberta I have been working to automate some of the QC's. I think pylinac is an awesome tool. One of the tests we perform is the calibration of the zero jaw position using the EPID or asymetric jaw test for overlapping fields. Essentially we measure the dose at the field intersection and find the gap/overlap between two adjacent fields. I didn't see any similar module in pylinac and I believe it could also be a neat new addition. I have worked a bit between courses to build a working prototype in python to replace an existing MATLAB tool and we are testing it. Please find enclosed a document explaining the procedure. I would like to know what are your thoughts and if it could be added to pylinac?

Best regards
Pedro
[QC_jaws.pdf](https://github.com/jrkerns/pylinac/files/2899051/QC_jaws.pdf)

jrkerns, 2019-02-25:
Hi Pedro, 
     Thanks for sharing. Couple of thoughts. 

1) If you want to leverage work already done you can use pylinac's [SingleProfile](https://pylinac.readthedocs.io/en/latest/core_modules.html#pylinac.core.profile.SingleProfile) class, which has several manipulation methods such as filtering, grounding, peak finding, etc. It looks like you have done this work already though, so maybe you're past that point
2) Based on my current understanding, I would not add it to pylinac. There are a few reasons. It's not as common as the other tests included in pylinac. MPC is a great tool for measuring jaw error and if you haven't validated it yet I would do that ASAP. Even with this, we all know the jaw is not as precise as the MLCs. I don't know if you are doing 3D or IMRT/VMAT, but for 3D I never used jaws to define field abutments for precisely this reason, even if I ran a test like the one you are doing; we always placed the MLCs 2mm into the field to be more confident in our field edge. For VMAT we would overlap the fields by a few cm (4-5) and use the optimizer to automatically ramp the doses up and down in the overlapping region, in which case the jaw accuracy doesn't come into play. I'm sure you guys have a valid reason to use the jaws but I am not familiar with it.

peterg1t, 2019-02-25:
Hi James,
Thanks for your promptly reply. If the MPC measures the jaw error you are totally right there is no need for this. To be honest we haven't fully finished implementing MPC's validation here (although we are almost there), that's why we keep QC of the jaws in our monthly QC.
Cheers,
Pedro

closed 2019-08-22 by jrkerns