jrkerns/pylinac #164: publish_pdf causes traceback in LeedsTOR

alanphys, 2019-02-25:
Hi James
Thanks for the updated LeedsTOR module. Results are looking much more sensible now. However the publish_pdf function is causing a traceback as follows:

>>> ld.publish_pdf()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/pylinac/planar_imaging.py", line 912, in publish_pdf
    self.save_analyzed_image(data, image=img, low_contrast=lo)
  File "/usr/local/lib/python3.6/site-packages/pylinac/planar_imaging.py", line 71, in save_analyzed_image
    self.plot_analyzed_image(show=False, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/pylinac/planar_imaging.py", line 883, in plot_analyzed_image
    Circle(center_point=self.phantom_center, radius=self.phantom_radius).plot2axes(img_ax, edgecolor='blue')
UnboundLocalError: local variable 'img_ax' referenced before assignment

**To Reproduce**
Steps to reproduce the behavior:
>>> ld = LeedsTOR('/home/alanphys/Programs/pylinac/pylinac/demo_files/planar_imaging/leeds.dcm')
>>> ld.analyze()
>>> ld.publish_pdf()

The problem is caused by line 883 in planar_imaging
       Circle(center_point=self.phantom_center, radius=self.phantom_radius).plot2axes(img_ax, edgecolor='blue')

Moving this statement to the if image: block solves the problem. 

If you are OK with this I will issue a pull request.

Regards
Alan

jrkerns, 2019-02-25:
> File "", line 1, in

Ah crap, yup. Artifact of testing. My unit tests seem to be having issues
with the PDF tests. Not sure why. Thanks!

jrkerns, 2019-02-25:
Fixed in 2.2.3 which was just pushed to PyPI.

closed 2019-02-25 by jrkerns