jrkerns/pylinac #165: Wrong unit in leaf RMS plot

keithoffer, 2019-02-28:
**Describe the bug**
As far as I can tell, there is conflicting units / wrong values in the leaf RMS plot compared to the rest of the reported numbers.

**To Reproduce**
1. Generate a PDF report for either trajectory logs or dynalogs
2. Observe the max RMS value at the top of the report (in mm)
3. Scroll down to the bottom of the report and observe the graph
4. Note the plot of RMS values (in mm) has the max value an order of magnitude smaller than that reported at the top of the report

Note that this behaviour is also seen in the plots generated by the trajectory log demo.

**Expected behavior**
The max RMS in the plot should be the same length as the one at the top of the report

**Screenshots**
From the trajectory log demo, areas of interest circled in red. Note that the code states 0.006 cm, the plot 0.006 mm.
[image]

**Additional context**
Tested with version 2.2.3, latest from pypi

Reports generated from demo files show the issue is common to both log types:
[demo_report_dyna.pdf](https://github.com/jrkerns/pylinac/files/2914370/demo_report_dyna.pdf)
[demo_tlog.pdf](https://github.com/jrkerns/pylinac/files/2914371/demo_tlog.pdf)

jrkerns, 2019-03-05:
Thanks for reporting! Let me take a look this afternoon.

jrkerns, 2019-03-05:
Confirmed. The plot is not corrected from cm (the default unit) to mm. Will convert all results to mm to be consistent when printing to console and PDF.

commit 27695790 (2019-03-10): Closes #165
closed 2019-03-10 by jrkerns