jrkerns/pylinac #166: 0 byte files crash dcmread

alanphys, 2019-03-01:
Hi James

I am having a minor issue with 0 byte files in the image directories when I use the CatPhan and Winston-Lutz modules. I use Aria over Citrix on Linux and for some reason it places a 0 byte .access^ file into the directory I export my images to. Dcmread attempts to read this file and because force is set to True tries to perform a seek past the beginning of the file.

I've created a workaround setting force=False in the dcmread statements and tweaking the error handling, but I am not sure what effect it will have on the rest of the project.

Should I list this as a bug on pydicom and rather sort the problem out there?

**To Reproduce**
On Linux open console
Create 0 byte file in image directory
$ touch mysillyfile.dcm
Start python and open CatPhan object
$ python3
>>> from pylinac import CatPhan604
>>> cp = CatPhan604('/path/to/image/directory/CBCT17012019')

On Windows
Open file explorer
Navigate to image directory
Right click, select "New", "Text Document" 
Run python and create CatPhan object as above.

Winston-Lutz module shows similar behaviour.

Traceback
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/pylinac/ct.py", line 887, in __init__
    self.dicom_stack = image.DicomImageStack(folderpath, check_uid=check_uid)
  File "/usr/local/lib/python3.6/site-packages/pylinac/core/image.py", line 986, in __init__
    if self.is_CT_slice(path):
  File "/usr/local/lib/python3.6/site-packages/pylinac/core/image.py", line 1019, in is_CT_slice
    ds = pydicom.dcmread(file, force=True, stop_before_pixels=True)
  File "/usr/local/lib/python3.6/site-packages/pydicom/filereader.py", line 880, in dcmread
    force=force, specific_tags=specific_tags)
  File "/usr/local/lib/python3.6/site-packages/pydicom/filereader.py", line 693, in read_partial
    fileobj.seek(-1, 1)
OSError: [Errno 22] Invalid argument

**Expected behavior**
pylinac should gracefully ignore all non-dicom files.

Regards
Alan

alanphys, 2019-03-05:
I think this can be solved more elegantly in pydicom and have created an issue (#42) there.
Regards
Alan

jrkerns, 2019-03-05:
Thanks for reporting this. Looks like the fix is in place for pydicom 1.3.

closed 2019-03-05 by jrkerns