jrkerns/pylinac #169: Log fluence error

jrkerns, 2019-03-11:
Fluence from Dynalogs and Trajectory logs seem to have an issue.

closed 2019-03-11 by jrkerns

Centrus007, 2019-04-08:
Hi James,

I'm no sure if you're already aware of this, but I thought I'd mention the following regarding this issue:

VMAT Dynalogs replace the dose fraction with gantry angle *by default*, **but the user can change this by editing the MLC controller startup script.** Specifically, the user can remove '#' preceding the _diagUseRapidArcDynalogFormat_ command.

I believe your original fluence calculations were fine for any Dynalogs for which the above process had been followed. We certainly noticed no issue, and we went to considerable effort validating your fluence maps.

I wonder whether it is better to just keep your original implementation, but include a recommendation for the user to perform the above process in the pylinac docs. Up to you, of course. :)

For confirmation, see page 41 of Varian's [DynaLog File Viewer 8.1 Reference Guide.pdf](https://github.com/jrkerns/pylinac/files/3052738/DynaLog.File.Viewer.8.1.Reference.Guide.pdf)

jrkerns, 2019-04-08:
Hi Matt, 
  Thanks for reporting this. I had overlooked that tidbit! All the VMAT dynalogs I have did not take advantage of that fact. 

I believe that if this change is enabled, i.e. it's back to using the relative 25000 dose index, then it should be fine (crosses fingers). See [here](https://github.com/jrkerns/pylinac/blob/master/pylinac/log_analyzer.py#L1646-L1654) where if the last index is 25000 no changes are made. 

If you provide a log like you described I would be happy to take a look at it.

Centrus007, 2019-04-08:
No worries! Glad to see both cases are handled anyway - I missed that.

If you'd like to take a look, here is [a zip](https://github.com/jrkerns/pylinac/files/3056609/RapidArc_dynalogs_with_dose_fraction.zip) containing a set of dlogs of a 2-arc VMAT delivery with dose fraction reporting enabled.

commit b3be1d85 (2019-03-11): Closes #169