jrkerns/pylinac #179: Change GUI from tkinter to Gooey

jrkerns, 2019-03-20:
The Tkinter interface is built into Python which means no added deps, but there seems to be no good way to test the GUI. Gooey provides an automatic conversion from cmd line to GUI. This means the GUI-specific testing is offloaded to a 3rd party library. This would require a rework of the current GUI, but add standardization, complex features (like progress bar, stdout pipe, etc), and by nature of its requirements a cmd line for pylinac would be added. There does not seem to be a use for the command line interface except for the directory analyzer, but this will give 2 benefits for the price of 1 and reduce the amount of untested code in the codebase.

alanphys, 2019-05-17:
Hmmm, interesting! I'm not entirely convinced that this will do what you want though. It is based on wxPython and so adds another dependency. At the end of the day you will still be left with a a fairly rudimentary GUI with little image support. I was toying with the idea of developing something using PySide2/QT5 based on a Dicom Image viewer but it is a lot of work. GUI's are by their nature difficult to test. Have you given some thought to perhaps divorcing the GUI component entirely from the code so that pylinac essentially becomes a library?

jrkerns, 2019-05-17:
Originally, I wasn't ever going to put a GUI inside. That's what the web app was for. But then I ran into limits on Heroku, ran out of time, and figured it was fine to throw something together in tkinter. Then it was all buggy (I had no way to test it programmatically), leading me to the present. I am fine keeping pylinac a library, I just had many requests from non-programmers trying to use it.

alanphys, 2019-05-20:
My apologies, I wasn't meaning to denigrate the GUI or the fantastic work you've done. The fact is that the GUI is extremely useful. Most of the people I know using pylinac use it via the GUI. What I think I'm trying to say is that if you are considering changing the GUI it may be worth investing some time and thought into the process and I am willing to assist. You have obviously been considering this given that you placed the post and I presume you were looking for some response.

jrkerns, 2019-05-21:
No offense taken! I'm all for a GUI, I just wanted something I could test, and like I said, at the time I was pushing for a web app. I still think that's a good idea, it's just not lightweight. I'm all for finding a better option. I also didn't pursue it more aggressively because I assumed a certain amount of adoption through QATrack+.

jrkerns, 2020-11-29:
Closed for the time being.

closed 2020-11-29 by jrkerns