jrkerns/pylinac #182: Pylinac tools for Elekta

zism, 2019-03-29:
Hi, I'm an Elekta user and I was wondering if you can incorporate tools for Elekta machine such as DynamicLogFiles, IVIEW EPID, XVI CBCT, Agility MLC...I can provide you data if needed.

bozhikovstanislav, 2019-08-02:
Hi . I do some research on DynamicLogFiles .  I can help you with doing the software working for your Elekta machine.  You can sent me the Dynamic LogFiles on bozhikov.stanislav@gmail.com if you want.

jrkerns, 2019-08-22:
Great. See this section https://github.com/jrkerns/pylinac#contributing. I would need the data and associated descriptions so that it can be parsed properly.

SimonBiggs, 2019-09-19:
@zism @bozhikovstanislav, the PyMedPhys library has some tools for Elekta log files (such as converting TRF files to CSV format https://pymedphys.com/user/interfaces/cli/trf.html). The API is still in flux and the library as a whole is still in beta, but you might get some benefit from it...

quantumquark07, 2020-05-26:
I just uploaded QA Images for two Elekta head types.
1. Agility Head on Versa HD (0.5cm leaves)
2. Synergy MLCi (1.0cm leaves)

Image Types Include:
1. FC2 Phantom (Rad Light)
2. QCKV - KV image Quality
3. QCMV - MV Image Quality
4. CBCT CATPHAN (Synergy one has a plug in the wrong place)
5. VMAT CAT - Static Gantry Picket Fence and Diaphragm Abutment Tests (Agility Versa HD Only)
6. VAMT PF - EPID sag causes image to skew (could do relative gamma on it)

Note that Elekta does not give the user the ability to apply dose or profile corrections to the EPID images.  Elekta leaves it up to the third party software to implement dose calibration and re-apply profile corrections.  

Worse yet the EPID images are automatically rescaled for better appearance to the human eye, and the rescaling factor (called "pixel factor") is not stored in the DICOM data.  The pixel factor is stored in the iView database.  Customers has to buy a special developer licence to allow them to use an API to search the iView database to get PixelFactor data for each image.

The nice thing about access to the iView database is that you can pull additional information about the beam and patient.  (See upload via contributing)

SimonBiggs, 2020-05-26:
Out of interest I didn't have much difficulty reading the iView database
with the "dbfread" library.

See the following notebook for a prototyping example:

https://www.github.com/pymedphys/pymedphys/tree/master/examples%2Fprotyping%2Fread_iViewDB%2F003_RCCC_batch.ipynb

quantumquark07, 2020-05-26:
That’s great news!

quantumquark07, 2020-05-26:
Do you have a newer iView with the upgrade to SQL?

SimonBiggs, 2020-05-26:
We have both, at different sites. I have been able to access both, but I
haven't yet put the time to port my code over to the new schema.

zism, 2024-01-24:
Hello,
I cannot launch the gui on Ubuntu 22.04. I'm getting the following message:
-> pymedphys gui
Traceback (most recent call last):
  File "/home/medphys/.local/lib/python3.10/site-packages/pymedphys/_vendor/apipkg/__init__.py", line 238, in __getattribute__
    return getattr(getmod(), name)
  File "/home/medphys/.local/lib/python3.10/site-packages/pymedphys/_vendor/apipkg/__init__.py", line 216, in getmod
    x = importobj(modpath, None)
  File "/home/medphys/.local/lib/python3.10/site-packages/pymedphys/_vendor/apipkg/__init__.py", line 103, in importobj
    module = __import__(modpath, None, None, ["__doc__"])
  File "/home/medphys/.local/lib/python3.10/site-packages/streamlit/__init__.py", line 70, in <module>
    from streamlit.delta_generator import DeltaGenerator as _DeltaGenerator
  File "/home/medphys/.local/lib/python3.10/site-packages/streamlit/delta_generator.py", line 90, in <module>
    from streamlit.elements.arrow_altair import ArrowAltairMixin
  File "/home/medphys/.local/lib/python3.10/site-packages/streamlit/elements/arrow_altair.py", line 27, in <module>
    from altair.vegalite.v4.api import Chart
ModuleNotFoundError: No module named 'altair.vegalite.v4'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/medphys/.local/bin/pymedphys", line 8, in <module>
    sys.exit(main())
  File "/home/medphys/.local/lib/python3.10/site-packages/pymedphys/cli/__init__.py", line 148, in pymedphys_cli
    args.func(args)
  File "/home/medphys/.local/lib/python3.10/site-packages/pymedphys/_gui.py", line 44, in main
    start.start_streamlit_server(streamlit_script_path, config)
  File "/home/medphys/.local/lib/python3.10/site-packages/pymedphys/_streamlit/server/start.py", line 26, in start_streamlit_server
    st.bootstrap.load_config_options(flag_options=config)
  File "/home/medphys/.local/lib/python3.10/site-packages/pymedphys/_vendor/apipkg/__init__.py", line 248, in __getattribute__
    raise ModuleNotFoundError(
ModuleNotFoundError:
                    PyMedPhys was unable to import "streamlit.bootstrap.load_config_options".
                    The easiest way to fix this issue is to use the "[user]"
                    option when installing PyMedPhys. For example,
                    with pip this can be done by calling
                    "pip install pymedphys[user]==0.39.3".