jrkerns/pylinac #183: Bakai Formula

bozhikovstanislav, 2019-04-16:
**Describe the bug**
A clear and concise description of what the bug is.

        # equation: (measurement - reference) / sqrt ( doseTA^2 + distTA^2 * image_gradient^2 )
        subtracted_img = np.abs(comp_img - ref_img)
        denominator = np.sqrt((doseTA / 100.0 ** 2) + ((distTA_pixels ** 2) * (grad_img ** 2)))
        gamma_map = subtracted_img / denominator

Here the denominator say doseTA/100.0**2 - This is not true --- it must be (doseTA/100.0)**2 
as it is in the formula

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jrkerns, 2019-04-17:
This gives the same result in Python:
>>>1/100.0**2
0.0001
>>>(1/100)**2
0.0001

bozhikovstanislav, 2019-04-17:
> What about
> 2/100.0 ** 2
> it give 0.0002
> (2/100.0)**2
> give 0.0004
> ?? I do not get it is doseTA alway 1
>
>
> Sent from Mailspring (https://link.getmailspring.com/link/2174E228-D24D-4348-997F-464FA0FB91AC@getmailspring.com/1?redirect=https%3A%2F%2Fgetmailspring.com%2F&recipient=cmVwbHkrMDA2NmI2YWUwNmI1MWI0MjZkYjI0Mzk3YTJjMjI1YWU2ZjUwY2MwODA5MzEwYmE2OTJjZWJhYzNkNTcyOTJhMTY5Y2UxOWRkZWU4ZkByZXBseS5naXRodWIuY29t), the best free email app for work

I am doing resurch on Gamma with a tests and
I do not like the results
I want to overide dostTA - because I want to use it .

Sent from Mailspring (https://link.getmailspring.com/link/2174E228-D24D-4348-997F-464FA0FB91AC@getmailspring.com/0?redirect=https%3A%2F%2Fgetmailspring.com%2F&recipient=cmVwbHkrMDA2NmI2YWUwNmI1MWI0MjZkYjI0Mzk3YTJjMjI1YWU2ZjUwY2MwODA5MzEwYmE2OTJjZWJhYzNkNTcyOTJhMTY5Y2UxOWRkZWU4ZkByZXBseS5naXRodWIuY29t), the best free email app for work

jrkerns, 2019-04-18:
I see now, yes for a value over 1 this is giving an incorrect value. The default is 1, so the default results should not change. Thanks for reporting

commit 33eb544d (2019-08-22): Closes #183.
closed 2019-08-22 by jrkerns