jrkerns/pylinac #187: `scipy.misc.imresize` removed from version 1.3.0 of SciPy.

Centrus007, 2019-05-20:
**Describe the bug**
`pylinac.core.image` uses `imresize` from `scipy.misc`, but `scipy.misc.imresize` [has been removed from SciPy as of version 1.3.0](https://docs.scipy.org/doc/scipy/reference/release.1.3.0.html?highlight=imresize#scipy-interpolate-changes).

A straight `pip install pylinac` will now pull `scipy==1.3.0`, which is now incompatible with sections of pylinac that use any of the (now removed) functions listed [here](https://docs.scipy.org/doc/scipy/reference/release.1.3.0.html?highlight=imresize#scipy-interpolate-changes).

**Proposed solution**
Consider replacing `scipy.misc.imresize` with `skimage.transform.resize` as per [the scipy docs](https://docs.scipy.org/doc/scipy-1.1.0/reference/generated/scipy.misc.imresize.html#scipy.misc.imresize), and similarly for any other uses of functions not present in version 1.3.0 of SciPy. Or pin to `scipy<=1.2.1`?

jrkerns, 2019-05-21:
Good catch, thanks.

Centrus007, 2019-05-22:
No worries :)

bozhikovstanislav, 2019-07-18:
Is someone have any idea where this module is placed ?

randlet, 2019-07-18:
According to the docs it was removed completely:
https://docs.scipy.org/doc/scipy-1.2.1/reference/generated/scipy.misc.imresize.html

However they give an alternative using Pillow directly.

Randy

commit 1df3432c (2019-08-22): closes #187
closed 2019-08-22 by jrkerns