jrkerns/pylinac #189: Trajectory log results are different on different computers

bozhikovstanislav, 2019-08-02:
**Describe the bug**
A clear and concise description of what the bug is.

The calculation are different on different computers .

There is problem in core directory file 
image.py 

change it with float32 to float64 fixed it

the numpy array must be of type  np.float64 !! 
.
.

40       img = load(path, dtype=np.float64)

.
.
.
624     img_x = spf.sobel(ref_img.as_type(np.float64), 1)
625     img_y = spf.sobel(ref_img.as_type(np.float64), 0)

this fixed it 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jrkerns, 2019-08-22:
Can you give me some more context here? What exactly is different and why is float64 consistent but float32 isn't?

bozhikovstanislav, 2019-08-22:
https://stackoverflow.com/questions/13421159/different-result-of-same-numpy-mean-calculation-on-two-computers?fbclid=IwAR2UJkbKazO4RSA8pE-aA4ZrCwwYN3annKmJOi47qbkVvNC9J8g4VjyEo4E

here is explained

bozhikovstanislav, 2019-08-22:
I found that when my professor is doing the same calculation on his computer . the results were differed than that of my computer . after changing np.float32 to np.float64 . everything go the same ...
as it was mention in the link above . it is related on how  numpy library is manipulating big set of array .... this is what I can say ..