jrkerns/pylinac #191: FlatSym

Zarita78, 2019-08-27:
I proceed as suggest on the documentation.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\ProgramData\Anaconda3\lib\site-packages\pylinac\flatsym.py", line 139, in analyze
    self.flatness = self._calc_flatness(flatness_method, vert_position, horiz_position)
  File "C:\ProgramData\Anaconda3\lib\site-packages\pylinac\flatsym.py", line 186, in _calc_flatness
    vert_flatness, vert_max, vert_min, vert_lt, vert_rt = flatness_calculation(vert_profile)
  File "C:\ProgramData\Anaconda3\lib\site-packages\pylinac\flatsym.py", line 19, in flatness_varian
    dmax = profile.field_calculation(field_width=0.8, calculation='max')
  File "C:\ProgramData\Anaconda3\lib\site-packages\pylinac\core\decorators.py", line 73, in wrapper
    return func(*args, **kwargs)
  File "C:\ProgramData\Anaconda3\lib\site-packages\pylinac\core\profile.py", line 490, in field_calculation
    return field_values.max()
  File "C:\ProgramData\Anaconda3\lib\site-packages\numpy\core\_methods.py", line 28, in _amax
    return umr_maximum(a, axis, None, out, keepdims, initial)
ValueError: zero-size array to reduction operation maximum which has no identity

https://github.com/jrkerns/pylinac/issues/152*

Hi,
I also have the very same problem using FlatSym with the latest version of pylinac.
I haven't found this issue with BeamImage in the past. I have tested it on the same images and I got the error above. It only seemed to work for one of the images but plots on the pdf appeared up side down.
Any advice would be appreciate it. Thanks very much.

keerthi-2019, 2019-08-30:
I also face the same issue with scanned film image(.tif)..help please

jrkerns, 2019-09-03:
Upload your image here and I will try to deduce. https://www.dropbox.com/request/YKRu4AmuPsXu55uQq761

keerthi-2019, 2019-09-10:
@jrkerns I have already send the images one week back.. didn't get any replies

randlet, 2019-09-10:
@keerthi-2019 A friendly reminder that sometimes it may take open source developers long periods of time before they can get to a specific issue you are having (if they ever get to it!).  It's not because they don't care, but just that often they have more pressing things they want/need to get done (like family, work, vacation etc).  Patience is required when asking people to volunteer their time to fix a problem you're having!

jrkerns, 2019-09-11:
Hi keerthi, of the two images you sent, the first was a tif file, and I will have to do some further digging here. The second file (dicom) can be analyzed by adding an .invert() call to the image before analyzing like so:

`fs = pylinac.FlatSym(...)`
`fs.invert()`
`fs.analyze(...)`

I will add some code to better detect this by default. Thanks

jrkerns, 2019-09-11:
I have a fix for the TIF image. I need to write some tests so it works for a range of images and will add it to the v2.2.8 release.

Zarita78, 2019-09-11:
I'm the one who uploaded the DICOM image.  It works perfectly now. Thanks a million!

commit 7221b08f (2019-09-19): Closes #191
closed 2019-09-19 by jrkerns