jrkerns/pylinac #194: Number of failed ROIs in LeedsTOR analysis does not match between analyzed image and plot

mchamberland, 2019-09-03:
**Describe the bug**
The number of failing low-frequency contrast ROIs shown on the LeedsTOR analyzed image does not match the number of failing ROIs on the plot of the low-frequency contrast.

**To Reproduce**
Run the LeedsTOR analysis on my uploaded image.

**Expected behavior**
The number of failing ROIs on the analyzed image should match the number of failing ROIs on the plot.

**Screenshots**
[image]

**Additional context**
I've uploaded my image to Dropbox.

mchamberland, 2019-09-03:
Hmm... actually, maybe I just don't understand what's being plotted!

I'm reading the source and documentation more carefully right now to try to make sense of those results.

mchamberland, 2019-09-03:
Okay, my confusion stems from the fact that the analyzed image shows the low contrast ROIs that fail the contrast threshold, whereas the plot is showing the contrast _constant_ values.

Given that the two other phantoms in the planar imaging modules use the contrast constant results on the analyzed image, would it make sense to also use the contrast constant for the Leeds phantom?

See [here for Leeds.](https://github.com/jrkerns/pylinac/blob/master/pylinac/planar_imaging.py#L868)

and compare to Las Vegas [here.](https://github.com/jrkerns/pylinac/blob/59d0055100345d809a8f85f6797587e43d594289/pylinac/planar_imaging.py#L256)

jrkerns, 2019-09-05:
This was a test and not a mistake in any way. 😑 You passed.

closed 2019-09-05 by jrkerns