jrkerns/pylinac #204: Pylinac 2.2.6 not calculating VMAT drmlc (and drgs?) correctly

tbezo, 2019-10-24:
With the new version of the VMAT Module the DRGS and DRMLC results are not consistent with the calculation of the values by hand (the way varian does the calculation). For the drmlc test I have one test with a failed segment which pylinac thinks is still passing.

[DR_MLC.zip](https://github.com/jrkerns/pylinac/files/3767496/DR_MLC.zip)

In the attached DR_MLC DICOM files the second segment should fail with a diff(x) of -3.72. Pylinac says the maximum deviation is 0.28 which is not what you get when you do the calculation by hand.

When I first tested the DRMLC module of pylinac (I think it was version 2.1) the values were the same as when calculated by hand.

jrkerns, 2020-04-10:
So this turned into a huge thing. I had to dig for quite some time to understand the effect of this across different scenarios. 

The long story short here is that for Varian machines with ARIA, the rescale intercept and slope were not being applied. I.e. it was the raw pixel values that were being compared. Given that the raw pixel values are not linear (not even close) this had the effect of lowering the deviation values (!!!).

The thing is, the DICOM behavior is inconsistent for different scenarios. For ARIA + TrueBeam the situation is described above and should be fixed immediately. For Mosaiq, it appears the raw values are already corrected so no changes are needed. For either older Mosaiq and/or Clinacs (The image is from several years ago so I don't know which one is the culprit) the rescale is present but incorrect. It is a simple linear rescale and 0 intercept. After applying the fix for the first case, this use case is now failing as the inversion detection gets confused due to vastly varying pixel response between the images. 

I'm still working on how to handle the last case without messing up the first. Will report back. This is alarming and I don't know how I didn't figure this out a long time ago. I think I was always just using pixel info for nearly everything else and figured it was appropriate. 

If you have VMAT images and can [donate them](https://www.dropbox.com/request/YKRu4AmuPsXu55uQq761) please do so and also specify the machine type and R&V system you are using along with version. I don't want others to be getting incorrect results so your donation can make sure this doesn't happen again.

tbezo, 2020-04-15:
I uploaded Testfiles (DR_GS and MLCsp) for our ClinacDHX, Truebeam 2.5 and our Unique. Clinac and Unique have the same Imager and corresponding Software Version. Our R&V is Aria 13.0. I hope that helps a bit.

commit 41dbb67e (2020-05-01): Closes #204
closed 2020-05-01 by jrkerns