jrkerns/pylinac #268: Apparent regression in BB finding algorithm

SimonBiggs, 2019-12-24:
**Describe the bug**
It appears that for some cases, from pylinac version 2.2.6 --> 2.2.7 there has been a regression in the BB finding algorithm. The screenshots given below compare the PyMedPhys wlutz algorithm with the pylinac algorithms from version 2.2.6 and 2.2.7.

**Expected behavior**
Expect a profile taken through the declared centre of the ball bearing should be symmetrical about the BB centre, therefore a flipped profile of the BB in the image should result in overlapping BB penumbras.

**Screenshots**
[image]
[image]
[image]

**To Reproduce**

To recreate the above screenshots run the following.

```bash
pip install pymedphys==0.17.1
```

```python
import matplotlib.pyplot as plt

import pymedphys
import pymedphys._wlutz.pylinac
import pymedphys._wlutz.iview
import pymedphys._wlutz.imginterp
import pymedphys._wlutz.findfield
import pymedphys._wlutz.findbb
import pymedphys._wlutz.reporting

image_path = pymedphys.data_path("pylinac_wlutz_regression_image.jpg")

penumbra = 2
edge_lengths = [20, 24]
bb_diameter = 8

x, y, img = pymedphys._wlutz.iview.iview_image_transform(image_path)

field = pymedphys._wlutz.imginterp.create_interpolated_field(x, y, img)
initial_centre = pymedphys._wlutz.findfield._initial_centre(x, y, img)

field_centre, field_rotation = pymedphys._wlutz.findfield.field_centre_and_rotation_refining(
    field, edge_lengths, penumbra, initial_centre
)    

bb_centre = pymedphys._wlutz.findbb.optimise_bb_centre( 
    field, 
	bb_diameter, 
	edge_lengths, 
	penumbra, 
	field_centre, 
	field_rotation, 
	pylinac_tol=0.5  # required to ignore diff to pylinac in this case
)

pylinac = pymedphys._wlutz.pylinac.run_wlutz(
    field, edge_lengths, penumbra, field_centre, field_rotation)

pymedphys._wlutz.reporting.image_analysis_figure(
    x,
    y,
    img,
    bb_centre,
    field_centre,
    field_rotation,
    bb_diameter,
    edge_lengths,
    penumbra,
)

plt.title('PyMedPhys Basinhopping Method')
plt.tight_layout()
plt.savefig('PyMedPhys.png')

pymedphys._wlutz.reporting.image_analysis_figure(
    x,
    y,
    img,
    pylinac['v2.2.6']['bb_centre'],
    pylinac['v2.2.6']['field_centre'],
    field_rotation,
    bb_diameter,
    edge_lengths,
    penumbra,
)

plt.title('Pylinac v2.2.6 Filter and Profile Method')
plt.tight_layout()
plt.savefig('Pylinac v2.2.6.png')

pymedphys._wlutz.reporting.image_analysis_figure(
    x,
    y,
    img,
    pylinac['v2.2.7']['bb_centre'],
    pylinac['v2.2.7']['field_centre'],
    field_rotation,
    bb_diameter,
    edge_lengths,
    penumbra,
)

plt.title('Pylinac v2.2.7 Filter and Scikit-Image Method')
plt.tight_layout()
plt.savefig('Pylinac v2.2.7.png')
```

**Additional context**
This is using a vendored version of pylinac so that the code can run pylinac 2.2.6 and 2.2.7 at the same time while comparing the results to what PyMedPhys has determined.

brjdenis, 2019-12-25:
Hi there!

Just adding something to this. I too have noticed that the BB finding algorithm may sometimes miss the geometric center of the BB for larger BBs (compared to the field), or when the contrast between the BB and the field is not large enough, or when I do the non-central WL test where the distribution of pixel values around the BB is always non-concentric.  Here is an example. On the left image with normal gray colormap it would seem that the BB center is positioned too much towards the right. But if one examines the distribution of pixel values, it turns out that the center of mass is indeed shifted outwards. On the right image I am using the "prism" colormap to show the pixel value distribution for the same image. The blue cross is the field CAX, the red cross is the BB center. I didn't do any further or better analysis, but looking at the image, I would say that the non-centrality of profiles is inherent to this algorithm. When I switched to version 2.2.7, I did some test and I didn't get results that were different by much more than 0.1 mm. I think I would get a similar result in older versions as well. But I didn't check it closely, so I can't be sure.

[image]

Live long and prosper!
Denis

SimonBiggs, 2019-12-26:
@brjdenis would you be able to upload those failing images to the PyMedPhys Zenodo file storage (https://zenodo.org/communities/pymedphys/)? I'd be keen to compare the WLutz results for those images to PyMedPhys' implementation.

brjdenis, 2020-01-05:
@SimonBiggs sure I can. I will upload the images that are similar to the one I showed on this thread.

brjdenis, 2020-01-06:
Hi again, I couldn't upload to zenodo. Here is instead an alternative link:
https://github.com/brjdenis/qaserver/blob/master/files/non-central-wl.zip

SimonBiggs, 2020-01-07:
Heya @brjdenis,

I have uploaded an example notebook analysing those images to the following link:

http://www.simonbiggs.net/wlutz

If you want to run that same code locally I have also attached the notebook file to this issue:
[wlutz-dicom-example.zip](https://github.com/jrkerns/pylinac/files/4030022/wlutz-dicom-example.zip)

The PyMedPhys wlutz code is still not officially released. It needs to be put through its paces first. But if you're keen on testing it and providing me more images when it struggles I would be massively appreciative of that.

Cheers,
Simon

brjdenis, 2020-01-07:
Hi @SimonBiggs,

I checked your results. It turns out that the difference between PyMedPhys and Pylinac calculated CAX-BB difference is less than 0.1 mm. I was honestly expecting it to be larger because to the naked eye it looks like the BB has the geometric center elsewhere. I guess I was wrong.

brjdenis, 2020-01-17:
Hi @jrkerns @SimonBiggs it's me again.

I repeated the analysis with Pylinac and PipsPro for several old images. The result is this: mostly they agree within 0.1 mm. However, in some cases, the difference can be larger. This plot shows the average difference between the two calculations (Pylinac minus PipsPro):

[image]

Images acquired with 10 FFF on Elekta VersaHD and high contrast 2.5 MV images on a Varian TrueBeam show the biggest difference when compared to PipsPro. I think that Pylinac's BB searching algorithm tends to fail for these images. I uploaded the example sets so that you can test them:

[elekta_BB_problem.zip](https://github.com/jrkerns/pylinac/files/4078661/elekta_BB_problem.zip)

[varian_BB_probleml.zip](https://github.com/jrkerns/pylinac/files/4078670/varian_BB_probleml.zip)

The results are:
[image]

For the three Elekta images, the differences are not very large, but are larger than 0.1 mm. And you can spot with the naked eye that Pylinac perhaps does not locate the center of the BB accurately. For Varian images, the results show a larger difference. For normal energies I don't see this, only for 2.5 MV.

Best regards,
Denis

brjdenis, 2020-01-17:
Sorry, I made an error in the second table.

[image]

SimonBiggs, 2020-01-19:
Just to clarify @brjdenis what version of pylinac was this?

brjdenis, 2020-01-19:
@SimonBiggs , the latest, 2.2.7

SimonBiggs, 2020-01-19:
Can you also repeat on version 2.2.6? Especially given this issue is to show that the previous version of `pylinac` is an improvement over the current.

brjdenis, 2020-01-19:
Unfortunately, I was randomly selecting images, so right now it would be hard to find them. But, I will repeat the analysis, and gather the images together for comparison.

SimonBiggs, 2020-01-19:
Yup, that'd be great. If you were keen I would also appreciate `pymedphys` to be included in that comparison if you could. It would be nice to have a representative comparison between pymedphys, pylinac 2.2.6, 2.2.7 and pipspro.

SimonBiggs, 2020-01-20:
@brjdenis, I have just undergone an initial release of some of the Winston Lutz code. I have written up an example using the DICOM images you provided earlier:

https://docs.pymedphys.com/howto/wlutz/varian-dicom-example.html

To run that example you will need to install pymedphys with:

```bash
pip install pymedphys==0.20.0
``` 

I would be very keen if you could run your tests against PyMedPhys as well.

brjdenis, 2020-01-23:
Hi @SimonBiggs  and @jrkerns !

I repeated the analysis with a new set of images (not the same as before). This time I collected all the data. I was not able to do the anaylsis with Pylinac 2.2.6 because of some problem with Scipy's imresize, and I also did not do the analysis with PyMedPhys because I didn't know how to set the parameters. 
But, all the data is in this zip archive, so you can redo the analysis yourselves. PipsPro results are included for each image. To run the analysis see calc.py.

https://drive.google.com/file/d/1AxpErc1IalboS6Sjo5EdRTJBJqZ1krsM/view?usp=sharing

I hope the link works.

Denis

jrkerns, 2020-04-11:
Hi guys, 
      I'm just now digging into this. The link works Denis. I tweaked it a bit to give me a 2D plot and now It's easier to find the images that have the largest deviations visually. Will report back. 
[image]

jrkerns, 2020-04-11:
Okay, so after some digging I found two issues. First let me describe roughly what pylinac was/is trying to do. 

To find the BB, pylinac uses a double-thresholding technique to find a [boolean pattern that looks like the BB](https://github.com/jrkerns/pylinac/blob/ad10a4835580a9e8e177962254d277e44ec078dc/pylinac/winston_lutz.py#L633-L652). Then it's [weighted by the intensity](https://github.com/jrkerns/pylinac/blob/ad10a4835580a9e8e177962254d277e44ec078dc/pylinac/winston_lutz.py#L655-L658) of the image. The idea is that the center of the BB should give more weight than peripheral pixels, as the boolean mask that's determined from step one isn't perfectly round and the centroid shouldn't be influenced by distal pixels with little intensity. The idea is to find a weighted centroid using an image like this: 
[image]

Unfortunately, that's not what was happening. The pixel weighting was using the convention that radiation increases the pixel intensity. This would result in the *lowest* pixel intensity at the center and the highest at the edges, which were choppy: 
[image]

Second, I finally determined that there was an issue with how the pixel values were being handled. As I recent discovered, the RescaleSlope and RescaleIntercept can 1) not exist, 2) be "dumb" values (1, 0), or 3) be correct with values that convert to CU/MU. Some of the images in Denis' set contained valid scale/intercept values while others did not. Images that do not have the tags or correct values end up having values with very, very small ranges. The edge of the image can be 15,900 and the radiation field values can be 16,200. This very small range of pixel values causes the intensity image to be essentially flat and practically boolean. This essentially means that when an image is using pixel values that the weighted intensity is not taken into account and we get an unweighted centroid. 

The following images show the agreement to Denis' dataset from PipsPro 1) before addressing this, 2) after correcting the inverted weighting intensity, and 3) after applying the correct intensity AND "stretching" images with raw pixel values to be normalized (out of field =~0 and in-field=~1). 

Before we started:
[image]

After intensity correction:
[image]

After intensity correction and weighting correction w/ raw pixels:
[image]

Pretty crazy right?

SimonBiggs, 2020-04-11:
@jrkerns that is some awesome work right there :) Thanks James :)

randlet, 2020-04-14:
Seconded! Very nice work James!

brjdenis, 2020-04-22:
This is great James. Thank you.

commit 19a94414 (2020-05-01): Closes #268
closed 2020-05-01 by jrkerns