jrkerns/pylinac #269: Winston-Lutz alternative for Elekta's EPID images

adabreug94, 2020-01-25:
**Elekta's EPID images aren't exported with gantry, collimator and couch angles**
As the pylinac's WinstonLutz module description states, Elekta's EPID images doesn't contain information about the angles at wich images are taken, no for collimator, nor couch, nor gantry.

**But we can include it in the image**
User could add a label to the image series through iView interface, wich handles EPID images acquisition in Elekta's linacs. This information will be saved in SeriesDescription tag of every .dcm file for that serie, so if we are planning to acquire images at different couch angles for same gantry angle, we could create a 'Treatment' called gantry# and add it some 'Treatment Fields' like couch60, couch300, etc. This information could be handled by the WinstonLutz module to extract information about parameters of each image used in the test.

jrkerns, 2020-04-07:
@alanphys What's your take on this?

alanphys, 2020-04-08:
Hi James

I think it is a possible workaround. If I am understanding David correctly you would create a plan with the field ID's reflecting the gantry, couch and collimator angles. I would use the same nomenclature as for the file names. When you export the EPIDs to dicom via the iView you can include additional information such as the series descriptor which should be automatically populated with the field ID.

I took a quick look at some old EPIDs. I'm not seeing the series description (Tag 0008 103E). I only see the study description (Tag 0008 1030) in some of them. Perhaps David could upload an example image series?

Unfortunately I am not working in an Elekta environment any more so I can't verify this. In the environment I was working in our EPID images were sent immediately from iView to Mosaiq and we would then export them from Mosaiq. I am not sure that this would work there.

Sorry I can't be of more help.
Regards
Alan

mchamberland, 2020-04-08:
I can confirm that the Series Description tag is properly filled out on recent versions of iView. I can upload sample (non-Winston-Lutz) images somewhere, if needed.

adabreug94, 2020-04-08:
Hi everyone. As was suggested, I add some sample images from a random day WL test for you consideration. Also I include an extra script which I use in here to automatically rename the dicom files attending on the SeriesDescription tag, edited in iView when the patient used for storing the test images was created. Hope it can be helpful

[Sample Files](https://drive.google.com/open?id=1TEYB19lAgQyqLpRnhRjIwX6SChwgMC4l)

alihm1987, 2020-04-11:
Hi everybody,

i am Ali. i am elekta distributer engineer in Iran and will be glad if i can do something to run this project better for Elekta machine. 
if you need some pictures or log files data from elekta machine with agility head please let me know. i will prepare them for you.

alihm1987, 2020-04-29:
Dear All,
here is my email address:
a.hosseini@fanavari.com
i want to promote this program for elekta machine.
Regards,

quantumquark07, 2020-05-26:
I just uploaded some Elekta images from a Synergy (MLCi 1cm) and a Versa HD (agility head 0.5cm) to the main upload link.  The uploads include recent CBCT's, WL images, FC2, QCKV, QCMV, PF tests, and VMAT PF images (kinda useless except for relative gamma analysis due to EPID sag).

Some notes about Elekta EPID's.

1. There is a rescaling factor hidden from you called the Pixel Factor which is only accessible via a direct query of the iView database.  The customer has to have a developer license to allow API calls to request the pixel factor (possibly collimator and gantry angles of each image too).  We currently employ this workaround for inhouse gamma software.

2. The iView workstation does not presently allow the user to set dose calibraiton and profile correction.  Calibrations and profile corrections are left to the third party application.