jrkerns/pylinac #270: Repeating LeedsTOR/QC3 analysis is adding to memory consumption

brjdenis, 2020-02-10:
Repeating the analysis of the LeedsTOR image 5 times, the memory consumption grows to 1 GB. Similar thing happens with QC3.

I am using Pylinac 2.2.7, Python 3.7.4 and Sypder 4.0.1 to run the script.

This may be expected behaviour. One can just clear the memory by hand in any case. But I thought I should report it.

```
from pylinac import planar_imaging
img = planar_imaging.LeedsTOR("myfile.dcm")
img.analyze()
```

randlet, 2020-02-11:
Not sure if it's the same issue, but a while ago I had an issue with excessive memory consumption and I think it turned out to be the lru_cache decorator used by many of the analysis routines.

jrkerns, 2020-02-11:
Yeah the modules aren't exactly memory-friendly =)

I thought it had to do with the lru_cache, but when I removed them it still gave me problems so I left them in. Maybe it was developer error in my testing strategy. Admittedly though, the lru_cache is quite helpful for those expensive calculations. 

In any event, yes, it's a known issue, but since it's a relatively uncommon event that you repeat analysis multiple times and combined with the benefit of the cache it's not been a high priority item thus far. 

Fixes welcome if you can prove it is (or isn't) the lru_cache.

commit d20ea3ad (2021-05-18): Fix memory leak!!! Closes #270
closed 2021-05-18 by jrkerns