jrkerns/pylinac #275: Upper limit pinning of numpy and scikit-image

SimonBiggs, 2020-04-15:
Hi @jrkerns,

Once https://github.com/jrkerns/pylinac/issues/268#issuecomment-613198392 lands I believe the best way forward will be for me to depend upon pylinac within PyMedPhys for my Wlutz algorithm. I still want to have an independent implementation with a significantly different approach at some point in the future, but current world events have adjusted priorities here.

Before having PyMedPhys depend on pylinac I was hoping to have the dependency upper limits removed from pylinac itself. So as to avoid downstream dependency hell I am aiming to pin packages as minimally as possible, instead allowing application developers to do their own pinning.

As such, I was hoping the following two upper limits may be able to be removed:
* https://github.com/jrkerns/pylinac/blob/ad10a4835580a9e8e177962254d277e44ec078dc/setup.py#L17
* https://github.com/jrkerns/pylinac/blob/ad10a4835580a9e8e177962254d277e44ec078dc/setup.py#L21

Any thoughts on your end?

Cheers,
Simon

jrkerns, 2020-04-15:
Yup. I know there was some reason for pinning it that way but I can't remember or find out =)

I updated Travis for the various versions and it should all be good there. Working on the next major version release

SimonBiggs, 2020-04-15:
Awesome thanks James :)

closed 2020-05-01 by jrkerns

SimonBiggs, 2020-05-01:
Thanks James :)

commit 15c08449 (2020-05-01): Bump to 2.3.0. Closes #275