jrkerns/pylinac #277: Ability to call WinstonLutz with just an image alone (not a DICOM object)

SimonBiggs, 2020-04-15:
Heya James,

Sorry for all these issues, given you have the hood of Pylinac + wlutz open I figured now would be the time to raise these. More than happy for you to not implement them, if all else fails I can continue to vendor pylinac instead of depending on it (would rather swap to a dependency though...).

One of the issues I had with using the Wlutz algorithm is I wanted to do it with a plain image file, not a DICOM file. To achieve this I slightly adjusted your WLImage object, see:

https://github.com/pymedphys/pymedphys/blob/019ee1cf62c9a65b6de6d0f614553d7b99fdaa37/pymedphys/_vendor/pylinac/winstonlutz.py#L61-L197

Once it was in that format I was able to wrap the object in a function and I was able to produce a functional API.

Here is my wrapper function, reading it now, I was really trying to do too many things at once and that should have been split up a bit more:

https://github.com/pymedphys/pymedphys/blob/019ee1cf62c9a65b6de6d0f614553d7b99fdaa37/pymedphys/_wlutz/pylinac.py#L33-L108

But essentially I was making a "run_wlutz" function, but within that function I was also rotating the image by the collimator angle so as to handle image rotation.

Anyway, as messy as it is, if you would like to use any of that code there that is wrapping pylinac feel free to use it under your own MIT license without needing to assign attribution in the header.

jrkerns, 2020-04-15:
Hi Simon, 
    No worries. Could you tell me more about why you need a normal image file vs a DICOM? I'm trying to stick to DICOM because it has all the metadata I need (e.g. SID, DPI) without having to pass it. I also didn't see a use case where a WL was done without an EPID.

SimonBiggs, 2020-04-15:
Yup, certainly. I am delivering a "Wlutz Arc" and I am recording the images in movie mode. That way I get about 300 frames spread over 360 degrees for about a 1 minute measurement. Unfortunately the Elekta Linac cannot export the movies as DICOM, so instead I am directly reading the image frames out of the database itself.

All I am really after is a function where I pass it an image and it spits out a bb_centre and field_centre in pixel coordinates. The input API could just be, pass numpy array with pixels values to a function, as well as an optional flag of whether or not to use the old or new algorithm (default to new).

SimonBiggs, 2020-04-15:
Here's a reasonable overview of the method:

https://github.com/pymedphys/pymedphys/pull/577

jrkerns, 2020-04-17:
Ah I see now, thanks. While I understand your use case, the methodology isn't wide-spread yet. If it becomes more mainstream then it would make sense to include it. To be honest I really wouldn't look forward to adding a lot more tests for this use case, so that's part of the reason =/ 

I can't comment on the coll rotation, but as for overriding, see this gist: https://gist.github.com/jrkerns/338d7210e620cd712f02bb73972ebf59

I tried to get cheeky by hacking the MRO, but it was a failure. It was much easier simply to replace the super call with setting the array. Note that almost all methods and attrs error out, but it does give the coordinates for field and bb.

SimonBiggs, 2020-04-17:
That object override should do the trick. I'll implement it that way once
3.0.0 lands.

jrkerns, 2020-04-30:
Working on it. Next version has several positives changes =)

closed 2020-04-30 by jrkerns

SimonBiggs, 2020-04-30:
> Working on it. Next version has several positives changes =)

Awesome! :) Thanks @jrkerns :)

SimonBiggs, 2020-06-01:
Heya @jrkerns was this fixed? Would you like me to still implement a work around in order to use plain images, or is there an API you would prefer me using?

Thanks :)

SimonBiggs, 2020-11-10:
Hi @jrkerns :),

Thanks for all the work in 2.3.*! The Wlutz is looking awesome thus far. Okay, so since there is not yet a public API interface for directly working with images in memory for the wlutz field and bb finding, I have continued to undergo some shenanigans.

...given the level of the nastiness of the shenanigans I would like to give you a heads up so that you don't feel like I am butchering all your work without at least letting you know...

So, I am using private API calls, and I understand that you're free to change those from one version to the next, so for the time being, PyMedPhys has been pinned to pylinac version 2.3.2:

https://github.com/pymedphys/pymedphys/blob/089fc131d9ac2e63ae47d4334ba92c9e64882396/pyproject.toml#L67

I have copied in a lot of your code, and made some adjustments, so I've done it under a license like the following, I hope that's okay:

https://github.com/pymedphys/pymedphys/blob/089fc131d9ac2e63ae47d4334ba92c9e64882396/lib/pymedphys/_vendor/pylinac/winstonlutz.py#L1-L16

I've made comments regarding what I've adapted here:

https://github.com/pymedphys/pymedphys/blob/089fc131d9ac2e63ae47d4334ba92c9e64882396/lib/pymedphys/_vendor/pylinac/winstonlutz.py#L18-L26

And I have given a heads up that this vendored/patched code should not be used in standalone mode, but instead its purpose is as a second check of the PyMedPhys algorithm:

https://github.com/pymedphys/pymedphys/blob/089fc131d9ac2e63ae47d4334ba92c9e64882396/lib/pymedphys/_vendor/pylinac/winstonlutz.py#L28-L39

To give you an idea of the patching I have done, here is an example of the new WLImage_2_3_2 class:

https://github.com/pymedphys/pymedphys/blob/089fc131d9ac2e63ae47d4334ba92c9e64882396/lib/pymedphys/_vendor/pylinac/winstonlutz.py#L63-L117

I hope this is all okay. Thank you so much for your work on this :). And potentially if enough usage of this PyMedPhys Wlutz Arc eventuates it might be worth upstreaming some of this code so that my shenanigans can cease :).