jrkerns/pylinac #278: CatPhan 604

AliWoWolx, 2020-04-21:
Hello,
I've been trying to use your very nice program with the 604 model of CatPhan.  I notice that if there are any slices in the CT dataset without the phantom in the error message "Unable to locate Catphan" is triggered on the np.max(edges) condition.  The code runs if the "empty" slices are removed and if the 504 module (and indeed the 503 and 600) is run on the same dataset, the error message is not triggered (though, of course, the results, aren't what one would hope for!).  Do you know why this is?
Thank you.

jrkerns, 2020-04-30:
Hmm, I'm not sure. If you [upload your images](https://www.dropbox.com/request/YKRu4AmuPsXu55uQq761) I'll take a look.

AliWoWolx, 2020-04-30:
Dear James,
That would be most kind, thank you.
I have submitted two data files.  They are of the 604 phantom, to the best of my knowledge.  The one labelled 183 has some CT slices that miss the phantom at either end and the one labelled Edited183 I manually removed the “blank” images.  My experience has been that the software will process the edited version but throws the first of the Unable to Locate Catphan error messages.  I haven’t been able to understand why this happens, particularly as when I tell the software that it’s the 504 (or 503 or 600) phantom, it will run with both datasets.
I am something of a python novice but am working with a more experienced researcher from UCL and an MSc student.  I am hoping that we will be able to automate our analysis of CBCT QA acquisitions and am tremendously grateful to you for this impressive piece of work.
Kind regards,
Alison
Alison Warry PhD MIPEM
Principal Physicist (Proton), UCLH

Radiotherapy Physics (PBT), 2nd Floor Central, 250 Euston Road, London, NW1 2PG
Tel: 020 3447 2225
alisonwarry@nhs.net<mailto:alisonwarry@nhs.net>
We are committed to delivering
top-quality patient care, excellent
education and world class research
safety  kindness  teamwork  improving
UCLH website<http://www.uclh.nhs.uk/> | Media<http://www.uclh.nhs.uk/aboutus/Media> | Become a member<http://www.uclh.nhs.uk/aboutus/FT/Pages/Becomeamember.aspx> | Support our charities<http://www.uclh.nhs.uk/aboutus/Campaignsandcharities> | Contact UCLH<http://www.uclh.nhs.uk/aboutus/Pages/ContactUs.aspx>
Find us on Facebook<http://www.facebook.com/UCLHNHS> | Follow us on Twitter<http://twitter.com/uclh> | Subscribe on YouTube<http://www.youtube.com/uclhvideo>

This message may contain confidential information. If you are not the intended recipient please inform the
sender that you have received the message in error before deleting it.
Please do not disclose, copy or distribute information in this e-mail or take any action in relation to its contents. To do so is strictly prohibited and may be unlawful. Thank you for your co-operation.

NHSmail is the secure email and directory service available for all NHS staff in England and Scotland. NHSmail is approved for exchanging patient data and other sensitive information with NHSmail and other accredited email services.

For more information and to find out how you can switch, https://portal.nhs.net/help/joiningnhsmail

jrkerns, 2020-05-12:
Alison, 
     As you mentioned, your datasets are not complete. The analysis is erroring out because the uniformity slice is not seen by pylinac. I'm not sure how you acquired these images but pylinac does not yet support partial analysis (i.e. skipping modules) but will in a future version. I suggest you take a full scan of the phantom.

AliWoWolx, 2020-05-13:
Dear James,

Thank you so much for looking at this for me – I had totally missed that the whole phantom was not in the image – sorry for wasting your time.

With best wishes,
Alison

jrkerns, 2020-05-13:
Not a problem. Good luck!

closed 2020-05-13 by jrkerns