jrkerns/pylinac #279: Obtaining deviations for each image during Winston-Luts test

mlanaret, 2020-04-27:
Dear,
Thank you for your very interesting program ! 
How could i obtain the values of the deviations for each axis and for each image analyzed ? in addition to a graph, in the . PDF report.
In addition, I cannot obtain an aesthetic report, my fields analyzed being very small (4mm conical collimator). Do you have a solution ?
Thanks for your help !

Cordially

jrkerns, 2020-04-30:
You will probably want the `cax2bb_vector`. That's the in-plane vector from field CAX to the BB. 

`wl = pylinac.WinstonLutz(...)`
`wl.images[0].cax2bb_vector  # will return a Vector w/ x/y/z components` 

To get the total deviation per axis there are [specific methods](https://pylinac.readthedocs.io/en/latest/winston_lutz.html#pylinac.winston_lutz.WinstonLutz.gantry_iso_size) for this, e.g. `wl.collimator_iso_size`. 

I'm not sure what you mean about the report. Can you attach your PDF here?

mlanaret, 2020-04-30:
Dear,
The problem on the pdf is page 2 with the EPID images, these are 7.5mm conical collimator fields.
I saw on the forum, the problem was recurrent and probably due to the very small sizes ed my fields.

Thank you for your answer, I would try your method to recover for each angulations, the deviation in the two axes of the center of the ball and not the vector between the two isocenters, that the quantities displayed in the graph "Relative Gantry displacement", i think. And put this numeric values on a tab in the pdf.

[WL.pdf](https://github.com/jrkerns/pylinac/files/4556649/WL.pdf)

Thanks you for all.

jrkerns, 2020-05-22:
Closed in https://github.com/jrkerns/pylinac/commit/4db07ea246584bb6f0a91e2fe2831b077ff0e76d

closed 2020-05-22 by jrkerns