jrkerns/pylinac #283: Las Vegas ROIs seem out of scale

mchamberland, 2020-05-08:
The Las Vegas analysis finds the right outline of the phantom but, when overriding the angle to account for the phantom orientation, the ROIs are not properly aligned, which suggests a scaling issue or perhaps different dimensions of my specific phantom model? I tried to find a phantom size that would align the ROIs properly, but haven't had luck.

**To Reproduce**
Analyze the image I have uploaded.

**Expected behavior**
If the phantom outline and angles are correct, I'd expect the ROIs to properly align with the phantom. The discrepancy seems to suggest my phantom has different dimensions than what is assumed in the code. Unless I'm missing something?

**Screenshots**
See attached. See also pictures of my phantom with scale.

**Additional context**
Acquired on Elekta iView. Note that Elekta documents state that you should acquire the image with a 12x12 field size, but doing so Pylinac cannot find the phantom (because the edges are cutoff). So, I've opened up the jaws, as recommended by Pylinac documentation.
[image]
[image]
[image]
[lasvegas.pdf](https://github.com/jrkerns/pylinac/files/4599431/lasvegas.pdf)

mchamberland, 2020-05-08:
It's a little easier to see how the ROIs don't quite line up on the PDF. Also, probably easier to do the analysis yourself and adjust window level.

jrkerns, 2020-05-08:
Interesting. I've never seen an LV phantom w/ screws in it before. Do you know who the manufacturer is or where you bought it?

jrkerns, 2020-05-08:
I'm used to something like this: https://www.researchgate.net/profile/Raymond_Schulz/publication/224957021/figure/fig1/AS:341234259382274@1458367995578/Phantom-images-acquired-on-NTX-for-image-quality-evaluation-a-a-Leeds-phantom-image.png

mchamberland, 2020-05-08:
Apparently, we got it from Elekta when we installed our current linacs in 2010.

I also had never seen the phantom with screws until I came to my current workplace. We had Varian linacs in my previous workplace.

jrkerns, 2020-05-08:
Elekta...

mchamberland, 2020-05-08:
I just went ahead and unscrewed the phantom. Turns out, it's assembled from metal sheets of varying thicknesses. There is one "column" of holes drilled into each sheet.

I've uploaded pictures [here](https://www.icloud.com/photos/#0aDekbaZU7CiCDD0icAB5IN2w).

jrkerns, 2020-05-08:
Never a dull moment! Pretty smart for a homemade hack.

crcrewso, 2020-05-11:
I'm not sure if you need to but I think you can get replacement nylon bolts for that threading.

mchamberland, 2022-01-11:
Pictures of the phantom [re-uploaded here](https://www.icloud.com/sharedalbum/#B15JEsNWnJuBvVj).

mchamberland, 2022-01-12:
None that I know of.

The iView customer acceptance document has a little diagram of the phantom, but it only shows you the diameter of the holes in each row. Nominally, they are 15 mm, 10 mm, 7 mm, 4 mm, 2 mm, and 1 mm. Not sure if that's useful to you.

mchamberland, 2022-01-12:
[iViewGT CAT version 4 03-25-09.pdf](https://github.com/jrkerns/pylinac/files/7856484/iViewGT.CAT.version.4.03-25-09.pdf)

See page 14 of the PDF.

jrkerns, 2022-01-13:
Ah ha! That's something. Thanks!