jrkerns/pylinac #289: CBCT HU Uniformity auto slice detectionn

crcrewso, 2020-05-19:
**Describe the bug**
Catphan 604 HU Slice not correctly picking center of module 

**To Reproduce**
Steps to reproduce the behavior:
1. How Do I provide you my dataset?
1. Run Catphan 604 analysis with defaults according to documentation

**Expected behavior**
The slice width wires should show in the center of the blue detection regions 

**Screenshots**
[image]
It's hard to see but the slice width wires are at the edges of the detection region, looks like the slice selected was too close to the air region of the phantom
[image]
This is consistent over a wide selection of our datasets. 

This issue is probably also contributing to our inconsistent measured slice width

jrkerns, 2020-05-20:
Yeah looks like an incorrect assessment of the central 404 slice. Upload your dataset here: https://www.dropbox.com/request/YKRu4AmuPsXu55uQq761

jrkerns, 2020-05-22:
So after looking at this dataset and a few others, the 604 appears to have a inconsistency with the other modules. The HU cylinders of the 604 are longer than the extent of the wire ramps. After stepping through the slices, the HU rois appears first, and then a few slices later the wire ramps start. 

Unfortunately, it's not a straightfoward fix. I can apply a correction to the HU slice position for 604 scans but it's pretty hacky and fragile.

crcrewso, 2020-05-24:
I would like to see two ways forward, could you post the hacky way for me to get through some clinical analysis that is somewhat time dependent, then we can get to testing the ultimate non-fragile solution for mainline integration.

jrkerns, 2020-05-24:
The hacky way is to modify this line of code:
https://github.com/jrkerns/pylinac/blob/master/pylinac/ct.py#L1026

You can either directly replace it with (based on my experience so far with
a few datasets):
hu_slices = np.array(hu_slices[2:])  # drop the first two indices where the
ROIs are present but not wire ramps
but this will change the behavior for all analyses. If you're using
multiple catphans then you'd need to add something like an if statement for
the 604 right above that line:
if isinstance(self, CatPhan604):
    hu_slices = hu_slices[2:]

crcrewso, 2020-05-24:
Great, Right now we're using just the one phantom. I'll confirm the thickness with the manufacturing specs. Do you think the following ultimate solution might be stable

if (is 604)
    STh = slice thickness
    index = inset distance / STh
    hu_slices = hu_slices[index:]