jrkerns/pylinac #295: Issue in FlatSym module

alessandrochiappini, 2020-05-20:
Hello,
after a flat install of pylinac with some additional modules when i use the demo of FlatSym, with this code:

**from pylinac import FlatSym
FlatSym.run_demo()**

obtain:

_runfile('C:/Users/brainfarm/Desktop/untitled0.py', wdir='C:/Users/brainfarm/Desktop')
Traceback (most recent call last):

  File "C:\Users\brainfarm\Desktop\untitled0.py", line 10, in <module>
    FlatSym.run_demo()

  File "C:\WPy64\python-3.8.2.amd64\lib\site-packages\pylinac\flatsym.py", line 129, in run_demo
    fs.analyze(flatness_method='varian', symmetry_method='varian')

  File "C:\WPy64\python-3.8.2.amd64\lib\site-packages\pylinac\flatsym.py", line 156, in analyze
    self.symmetry = self._calc_symmetry(symmetry_method, vert_position, horiz_position, vert_width, horiz_width)

  File "C:\WPy64\python-3.8.2.amd64\lib\site-packages\pylinac\flatsym.py", line 233, in _calc_symmetry
    vert_profile = self._get_vert_profile(vert_position, vert_width)

  File "C:\WPy64\python-3.8.2.amd64\lib\site-packages\pylinac\flatsym.py", line 215, in _get_vert_profile
    left_width = int(round(self.array.shape[1]*vert_position - self.array.shape[1]*vert_width/2))

AttributeError: 'FlatSym' object has no attribute 'array'_

jrkerns, 2020-05-22:
Ah, this was a dumb move on my part. I merged #200 into master but then didn't checkout 2.3.1 in order to make the bump to 2.3.2. Reverting to 2.3.0 or .1 should fix this for now. 2.4 is also due out soon and will contain this fix. Thanks

closed 2020-05-22 by jrkerns

myaqoub, 2020-06-22:
Hi,
I'm running Python 3.7 within jupyter and having exactly the same problem. how to fix it please?
Thank you!

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-21-0bda45c5e24c> in <module>
----> 1 my_img.analyze(flatness_method='varian', symmetry_method='varian', vert_position=0.5, horiz_position=0.5)

C:\ProgramData\Anaconda3\lib\site-packages\pylinac\flatsym.py in analyze(self, flatness_method, symmetry_method, vert_position, horiz_position, vert_width, horiz_width)

C:\ProgramData\Anaconda3\lib\site-packages\pylinac\flatsym.py in _calc_symmetry(self, method, vert_position, horiz_position, vert_width, horiz_width)

C:\ProgramData\Anaconda3\lib\site-packages\pylinac\flatsym.py in _get_vert_profile(self, vert_position, vert_width)

AttributeError: 'FlatSym' object has no attribute 'array'

commit 7afb5fb5 (2020-05-22): New flatsym.py behavior w/ profile widths. Also closes #295