jrkerns/pylinac #296: Profile parameter calculator

crcrewso, 2020-05-20:
**Is your feature request related to a problem? Please describe.**
No issues - Functionality extension

**Describe the solution you'd like**
- The ability to digest calibrated profile datasets from IC Profiler or equivalent device. 
- Calculate Flatness/Unflatness, symmetry, beam center, penumbra width, etc from calibrated profiles

**Describe alternatives you've considered**
- Currently individual metrics are manually entered into QATrack. 
- Could Implement directly in QATrack
- Could create an extension library 

**Additional context**
- I will be implement this functionality, if it's of interest to this community, my implementation could be pylinac based and pushed upstream. If there's no community interest then it makes sense to create the features directly in qatrack.

jrkerns, 2020-05-22:
I wrote a file parser example [here](https://gist.github.com/jrkerns/12ab58f5fc328a6f2cb8fdcbfabb95c3) a little while ago. I think it would indeed be pretty useful to have such a parser. Feel free to make a pull request against pylinac. You are also free to do it on your own if you'd like. I'm a slow dinosaur and it takes me a while to get changes out so if you want it asap then probably release on your own, otherwise I'm happy to review and work on this. The good news is this should be a pretty basic implementation.

crcrewso, 2020-05-24:
I  can see what you're doing there. It's a great start! My first guess for a layout is:
- A reader class that selects the parser and returns the calibrated profiles for analysis in a consistent way, ie. inplane, crossplane, etc, based on metadata. 
- An analysis class that takes in pylinac profiles and the analysis method of interest. 

Does this make sense? Would you rather a different organization method? If it does I'll work on getting the feature into a PR.

jrkerns, 2021-04-06:
Partially closed in #361. The integration with flatsym will happen after integrating #331

crcrewso, 2021-04-06:
For testing, what test cases would you like to see added? I can provide (or generate) a few on our IC Profiler for Unit Testing

jrkerns, 2021-04-07:
That would be much appreciated. I have the typical data of open beams. IIRC there was a movie-like mode; I don't have any of that data. I also don't know if anything has changed in the firmware/file format since my data was taken (~2.5 years ago). Also in the file data it says a detector may be ignored. It's not clear if that's a user-directed thing, or something SNC does on the backend somewhere. If any of that rings a bell then that data would be appreciated to fill out the parser. Thanks!

crcrewso, 2021-04-07:
I don't have movie mode or live mode, but, that's only really useful for beam alignment and the output of the first MU's 

I'll get a dataset together with dose and array calibrations, flat fields, excluded detectors, etc. While I'm putting that data together would you like some PD's of asymmetric jaw fields? [AAPM recommendation](https://aapm.onlinelibrary.wiley.com/doi/pdf/10.1002/acm2.12080) monthly test 8, [CPQR ML14](https://dx.doi.org/10.1002%2Facm2.12236) or is this outside the scope of the flatness/symmetry module

jrkerns, 2021-04-07:
Re movie mode: I agree, I just don't have that data. I'm not keen on adding
it, but you asked =)

By PD do you mean EPID images? Asymmetric fields are not outside the scope
of the flat/sym module and so such images are welcome. TBH I'd be more keen
on asymmetric MLC fields since the penumbra will be different. Thus far
I've used either the image generator module or Portal Dosimetry fluence
generation but you can never beat the real thing. Alan did an overhaul of
the flat/sym module with better interpolation for low density data (e.g.
Profiler) but I haven't gotten to it yet (read: lazy). I haven't see how
well it does for that low density data. After that, verifying asymmetric
data would be a good idea.

crcrewso, 2021-04-07:
Yes, I did mean EPID. I'll see about grabbing the same asymmetric MLC and jaw fields on EPID and IC Profiler. We've retired our profiler2 so I won't be able to go back that far.