jrkerns/pylinac #298: CatPhan Regression - Runtime Warning

crcrewso, 2020-05-24:
**Error Message**
/home/crcrewso/venv/pylinactest/lib64/python3.8/site-packages/numpy/core/fromnumeric.py:3334: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/home/crcrewso/venv/pylinactest/lib64/python3.8/site-packages/numpy/core/_methods.py:161: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
/home/crcrewso/venv/pylinactest/lib64/python3.8/site-packages/numpy/core/_methods.py:216: RuntimeWarning: Degrees of freedom <= 0 for slice
  ret = _var(a, axis=axis, dtype=dtype, out=out, ddof=ddof,
/home/crcrewso/venv/pylinactest/lib64/python3.8/site-packages/numpy/core/_methods.py:185: RuntimeWarning: invalid value encountered in true_divide
  arrmean = um.true_divide(
/home/crcrewso/venv/pylinactest/lib64/python3.8/site-packages/numpy/core/_methods.py:209: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)

**Describe the Bug** 
Warning from numpy about mean of empty slice 

**To Reproduce**
Steps to reproduce the behavior:
1. Install master version of pylinac with currently listed dependancies 
1. run CT analysis on Dataset 
1. See error
1. Analysis continues with what appears to be accurate results

**Expected behavior**
No warning message

jrkerns, 2020-06-29:
You can filter them (e.g. https://stackoverflow.com/questions/29347987/why-cant-i-suppress-numpy-warnings) or upgrade numpy.

jrkerns, 2021-04-05:
See if Randy's fix removes these for you: https://github.com/jrkerns/pylinac/pull/355.