jrkerns/pylinac #301: Starshot issue

kpaenov, 2020-06-22:
Greetings, I'm a bit new to the project and have been trying unsuccessfully to perform a Starshot analysis using multiple dicom images acquired on a Varian Halcyon v1.0 EPID.
This run was performed on a Ubuntu 16.04 with python3.6.9 and e venv for the pylinac, I've also tried running on anaconda with the same result. The same test set of images was successfully analyzed by a colleague of mine using pylinac at his center.

The script I'm trying to use looks something like this 

from pylinac import Starshot

star_imgs = [
                "./films/RI.QA_MV_0_0a.dcm",
                "./films/RI.QA_MV_0_0a1.dcm",
                "./films/RI.QA_MV_0_0a2.dcm",
                "./films/RI.QA_MV_0_0a3.dcm"
        ]

mystar = Starshot.from_multiple_images(star_imgs)
mystar.analyze()

And i get this error message

>>

/home/sysadm/.local/lib/python3.6/site-packages/numpy/core/_methods.py:160: RuntimeWarning: overflow encountered in reduce
  ret = umr_sum(arr, axis, dtype, out, keepdims)
/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/image.py:547: RuntimeWarning: invalid value encountered in double_scalars
  dist_to_5 = abs(p50 - p5)
/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/image.py:548: RuntimeWarning: invalid value encountered in double_scalars
  dist_to_95 = abs(p50 - p95)
/home/sysadm/.local/lib/python3.6/site-packages/numpy/lib/function_base.py:1280: RuntimeWarning: invalid value encountered in subtract
  a = op(a[slice1], a[slice2])
Traceback (most recent call last):
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/profile.py", line 237, in _get_initial_peak
    initial_peak = initial_peak_arr[0]
IndexError: index 0 is out of bounds for axis 0 with size 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "starshot.py", line 11, in <module>
    mystar.analyze()
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/decorators.py", line 73, in wrapper
    return func(*args, **kwargs)
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/starshot.py", line 214, in analyze
    start_point = self._get_reasonable_start_point()
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/starshot.py", line 157, in _get_reasonable_start_point
    fwxm_x_point = SingleProfile(x_sum).fwxm_center(80) + left_third
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/profile.py", line 371, in fwxm_center
    fwxm = self.fwxm(x, interpolate=interpolate)
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/profile.py", line 359, in fwxm
    li = self._penumbra_point(LEFT, x, interpolate)
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/decorators.py", line 73, in wrapper
    return func(*args, **kwargs)
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/profile.py", line 275, in _penumbra_point
    peak = copy.copy(self._initial_peak_idx)
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/profile.py", line 214, in _initial_peak_idx
    x_idx = self._get_initial_peak(self._passed_initial_peak)
  File "/home/sysadm/.local/lib/python3.6/site-packages/pylinac/core/profile.py", line 243, in _get_initial_peak
    raise ValueError("A reasonable initial peak was not found in the profile. Ensure peak is not at profile edge")
ValueError: A reasonable initial peak was not found in the profile. Ensure peak is not at profile edge

Svobodo, 2020-10-06:
Hello,
I have exactly the same issue.
Any update?
Regards
PT

kpaenov, 2020-10-06:
> Hello,
> I have exactly the same issue.
> Any update?
> Regards
> PT

Greetings the author of the project contacted me a few weeks ago to request a test set of my images and I haven't gotten a replay from him since. 
Sadly after loosing a day or two trying to figure the issue on my on, I've given up on the Starshot module

Svobodo, 2020-10-17:
Hello,
I worked around the problem making manually an image sum of my DICOMs using imageJ (https://imagej.nih.gov/ij/).
Importing just the resulting TIF image the module is working.
Regards

jrkerns, 2020-11-28:
This is fixed in v2.4

closed 2020-11-28 by jrkerns