jrkerns/pylinac #302: FlatSym problem

myaqoub, 2020-06-23:
Hi,
Im trying to follow the same steps in tutorial to use FlatSym, I used:

```my_file = r"C:\xxxxx.dcm"
my_img = FlatSym(path=my_file)
my_img.analyze(flatness_method='varian', symmetry_method='varian', vert_position=0.5, horiz_position=0.5)```
but got this error:
`ttributeError                            Traceback (most recent call last)
<ipython-input-48-0bda45c5e24c> in <module>
----> 1 my_img.analyze(flatness_method='varian', symmetry_method='varian', vert_position=0.5, horiz_position=0.5)

C:\ProgramData\Anaconda3\lib\site-packages\pylinac\flatsym.py in analyze(self, flatness_method, symmetry_method, vert_position, horiz_position, vert_width, horiz_width)

C:\ProgramData\Anaconda3\lib\site-packages\pylinac\flatsym.py in _calc_symmetry(self, method, vert_position, horiz_position, vert_width, horiz_width)

C:\ProgramData\Anaconda3\lib\site-packages\pylinac\flatsym.py in _get_vert_profile(self, vert_position, vert_width)

AttributeError: 'FlatSym' object has no attribute 'array''

I tried this change:
'''my_img = FlatSym.open_path=(my_file)'''
but got this error:

```AttributeError                            Traceback (most recent call last)
<ipython-input-50-0bda45c5e24c> in <module>
----> 1 my_img.analyze(flatness_method='varian', symmetry_method='varian', vert_position=0.5, horiz_position=0.5)

AttributeError: 'str' object has no attribute 'analyze'``

not sure what I'm doing wrong here?

jrkerns, 2020-06-29:
Downgrade to 2.3.1 and you should be fine. 2.3.2 has a bug in the flatsym module.

crcrewso, 2020-07-13:
I'm trying to extend FlatSym to also be able to handle SNC IC Profiler data. I can see a couple ways to do this. 
1. Convert the profiler data to an array of floats with 0's for all non-measured cells
1. Leave each profile as individual axes and create a new array init for each file format/data type 

Is there a fix for the 2.3.2 regression? it's getting in the way of my testing.

alanphys, 2020-11-04:
Hi James

Did my changes introduce this? I'll try and take a look.

Regards
Alan

alanphys, 2020-11-09:
OK, I see this has been fixed.

closed 2020-11-28 by jrkerns