jrkerns/pylinac #305: invert parameter to `LeedsTOR.analyze()` does not always force inversion

randlet, 2020-08-05:
Calling `LeedsTOR.analyze(invert=True)` does not always result in the image being inverted.

After the recent refactoring of the planar_imaging module, the `ImagePhantomBase.analyze` method looks like:

```
def analyze(self, ..., invert=False, ...):
    ...
    self._check_inversion()   # not implemented in LeedsTOR
    if invert:
        self.image.invert()
    self._preprocess()
```
however, LeedsTOR does not implement `_check_inversion` itself and instead checks the inversion in `_preprocess`:

```
    def _preprocess(self) -> None:
        self.image.check_inversion_by_histogram()
        if self._is_counter_clockwise():
            self._flip_image_data()

```

since the inversion is checked again (and hence potentially inverted)  in `_preprocess`, it can undo users request for `invert=True`.  

I think the simple fix is just to move the `self.image.check_inversion_by_histogram` call to `LeedsTOR._check_inversion`.  PR incoming.

If you want a sample file for this case let me know and I'll ask permission to share.

randlet, 2020-08-05:
Also noticed there is now no inversion check for the SIQC3 phantom.  Was this intentional or is it missing a:

```
    def _check_inversion(self) -> None:
        self.image.check_inversion_by_histogram()
```

as well?

brjdenis, 2020-09-06:
It seems to me that the culprit is the order of functions between lines 154 and 157. The _preprocess function changes the image back after inversion.

randlet, 2020-09-06:
@brjdenis indeed.  Fix is here https://github.com/jrkerns/pylinac/pull/306

closed 2020-11-20 by jrkerns