jrkerns/pylinac #307: Results summary for QAtrack and similar programs

crcrewso, 2020-08-25:
**Is your feature request related to a problem? Please describe.**
As the code base refactors, the calls to access specific measurements changes, creating a lot of surprise rework when updating. This has been a significant hurdle in trying to implement the CatPhan analysis through QAtrack+. 

**Describe the solution you'd like**
I would be happy to implement:
A single dictionary containing all measurement values would be returnable from a CBCT object, this dictionary would have strings as keys and primitives (or strings) as values. 
    i.e. myCBCT['Air'] = HU of air cavity of the CatPhan. 

**Describe alternatives you've considered**
If this is not advisable then a consistent API should be implemented for accessing all results

crcrewso, 2020-09-01:
First attempt at solving this is up  https://github.com/crcrewso/pylinac/commit/113824c5ca5c7b6f453cdf2c4b704c3ce2398d74

randlet, 2020-11-24:
Thanks for proposing this @crcrewso .  Having a more stable API for accessing results so you don't need to worry about things like the recent mtf api change would indeed make updating pylinac versions nicer for 3rd party applications.  I will try to pitch in with some of the other modules as time (lol) permits.

commit cca3dc8d (2021-04-06): Closes #307 for all except flatsym.
closed 2021-04-06 by jrkerns