jrkerns/pylinac #312: CatPhan Find Origin Slice

4Dct, 2020-09-22:
**Describe the bug**
Calling a set of CATPHAN503 images result in the failure to find the origin slice
"ValueError: No slices were found that resembled the HU linearity module"

**To Reproduce**
from pylinac import CatPhan503
cbct_folder = "mypath+file"
mycbct = CatPhan503(cbct_folder)
mycbct.analyze()

**Expected behavior**
the images are noisy (streak artifact) but the CTP404 module is visible (see attachment) and should show a circle profile contains both low (<800) and high (>800) HU values

**Screenshots**
CTP404 module slice
[image]

thanks for your help, and pylinac which is great

crcrewso, 2020-09-23:
There are quite a few issues with this image that would prevent the pylinac module from detecting it. The high  noise to contrast ratio would mess with the hu's and I can't even see the wires used for slice width. Do any of your other tools detect this as a ctp404 slice?

4Dct, 2020-09-28:
Thanks for that, I have commercial software where the ctp404 slice can be manually selected.

Can the find origin slice be easily set manually. For the majority of the datasets I have, the keyslice can be found with pylinac.

crcrewso, 2020-09-28:
Not at present, there's a lot of autodetection going on that tries to make it unnecessary. That panel is either in pretty bad shape or you're underdosing for the scan mode. I would say that such analysis is far enough from standard to require manual analysis with your existing solution.

closed 2020-09-29 by 4Dct