jrkerns/pylinac #315: TrajectoryLog resolution

Svobodo, 2020-09-28:
**Describe the bug**
Changing resolution setting gamma percent scores changes in an inconsistent way.
Reducing resolution from 0.01 from to 0.027 mm the gamma score reduces its values. After this, slightly decreasing resolution form 0.027 to 0.028 the gamma score jumps to about 100% (see attached graph)
[image]
[log.zip](https://github.com/jrkerns/pylinac/files/5292792/log.zip)

Is there an explanation for this behavior?

The test was performed on the attached log.bin from a Varian TrueBeam Linac (attached).
Other Settings:
- distTA = 0.01 mm
- version 2.3.1 and 2.3.2

Regards
PT

Svobodo, 2020-10-01:
Any comment on this?
Regards
PT

jrkerns, 2020-11-20:
Interesting. Haven't noticed this behavior before. This will require further investigation it seems.