jrkerns/pylinac #323: SingleProfile._penumbra_point may return incorrect points for some fields

randlet, 2020-11-12:
When analyzing some DRGS images a client encountered a bug in the penumbra detection routine which they figured out is a problem with the existing SingleProfile._penumbra_point algorithm.  The current algorithm [starts at the middle of the profile and works its way out](https://github.com/jrkerns/pylinac/blob/14a5296ae4ee0ecb01865d08f15070c82e19fc45/pylinac/core/profile.py#L289), stopping when it encounters a profile value below the threshold value.  However, this algorithm may get terminated prematurely if there is a dip in the profile that falls below the threshold even if it is not at the field edge.  For example see this image where there is a dip to ~50% prior to reaching the field edge (I've added dotted lines where Pylinac currently considers the field to be):

[image]

A fix is to instead use e.g. np.where to detect the outermost points that cross the threshold value which results in detected field like this:

[image]

PR incoming.

jrkerns, 2020-11-20:
This is correct. To be fair, that's why I called the class SingleProfile. It doesn't claim to work for these types of images. It was originally designed only for open fields.

jrkerns, 2020-11-20:
I do like your fix idea though. That would add some robustness, generally speaking.

randlet, 2020-11-20:
Fair enough.  The fix worked well but broke LeedsTor, I think, because it implicitly relies on the penumbra finding going from inside out.

alanphys, 2020-11-30:
I need to point out that this method may also run into problems if you have artefacts in your image such as burn pixels, location markers or writing.

Regards
Alan

alanphys, 2020-12-02:
For what it's worth this modification of Randlet's fix will give all points that cross the threshold:
`crossings = np.where(np.diff(np.sign(y_data - threshold)))[0]`

You can then do:
`        if len(crossings) > 1:`
`            raise IndexError("The profile has more than one point that corresponds to the threshold")`
`        elif len(crossings) == 0:`
`            raise IndexError("The point of interest was beyond the profile; i.e. the profile may be cut off on the side")`
`        else:`
`            rest of code...
`

jrkerns, 2021-02-09:
Closed as scipy's implementation was chosen.

randlet, 2021-04-15:
Unfortunately the current implementation still suffers from the same problem!

[image]

Trying to figure out a way to fix this now.

randlet, 2021-04-15:
I could be wrong, but after reviewing, I don't think it's actually possible to use find_peaks in a case like this to get the fwxm (scipy is always going to consider the extent of the peak to be the innermost 50% points).   That doesn't mean it's not working as intended for SingleProfiles, but I think it does make it unsuitable for calculating the fwxm_center for DRGS/DMLC fields like these.

randlet, 2021-04-15:
Sorry to spam the comments here, but given that the DRGS/DMLC only needs the fwxm_center and not the fwxm, it might be better in that case to use a MultiProfile and then use the location of the middle peak (or something similar).  Too late for me to investigate tonight though.

jrkerns, 2021-04-15:
Crap on a cracker! I guess it will indeed need a different approach or some parameter tweaking. The problem with using a middle peak is that we cannot assume there will be an odd number of peaks (assuming each ROI is a peak. If we simply use all data there is a valley in the true center with an even number).

randlet, 2021-04-15:
Yeah, I realized that this morning while looking at some other data.

jrkerns, 2021-04-15:
Is this still open given #370 ?

randlet, 2021-04-15:
Yes.  I'll get you a minimal reproduction in a minute.

randlet, 2021-04-15:
```python
import numpy as np
import matplotlib.pyplot as plt
from pylinac.core.profile import SingleProfile
prof_data = np.array([ 0.01491,0.00923,0.01014,0.00758,0.00601,0.00335,0.00499,0.00049,0.00440,0.00402,0.00000,0.00237,0.00369,0.00449,0.00254,0.00323,0.00430,0.00932,0.00675,0.00659,0.00565,0.00553,0.00197,0.00548,0.00467,0.00485,0.00669,0.00377,0.00486,0.00390,0.00821,0.00256,0.00302,0.00594,0.00576,0.00749,0.00459,0.00563,0.00529,0.00729,0.00598,0.00481,0.00701,0.00632,0.00648,0.00511,0.00915,0.00826,0.00668,0.00508,0.00872,0.00468,0.00520,0.00875,0.01141,0.01115,0.00681,0.00760,0.00584,0.00878,0.00574,0.01002,0.01081,0.00754,0.01161,0.01091,0.00690,0.01044,0.00790,0.01113,0.01180,0.00843,0.01019,0.00738,0.01068,0.01135,0.01197,0.00908,0.01025,0.01529,0.01267,0.01253,0.01415,0.01333,0.01219,0.01264,0.01482,0.01424,0.01098,0.01629,0.01180,0.01586,0.01424,0.01613,0.01562,0.01615,0.01711,0.01476,0.01607,0.01530,0.01485,0.01435,0.01750,0.01474,0.01505,0.01811,0.01979,0.01779,0.01567,0.02049,0.01814,0.01950,0.01905,0.01757,0.02093,0.01992,0.01942,0.01786,0.02251,0.02197,0.02363,0.02368,0.02345,0.02310,0.02316,0.02121,0.02285,0.02418,0.01962,0.02261,0.02775,0.02388,0.02274,0.02535,0.02336,0.02517,0.02545,0.02637,0.02847,0.02220,0.02618,0.02763,0.02770,0.02729,0.02751,0.03110,0.03061,0.02728,0.02963,0.03049,0.03677,0.02982,0.03031,0.03385,0.03327,0.03113,0.03840,0.03169,0.02959,0.03135,0.03165,0.03234,0.03059,0.03162,0.03333,0.03549,0.03593,0.03452,0.03673,0.03975,0.04010,0.03778,0.03860,0.03541,0.04154,0.04526,0.04346,0.04022,0.04357,0.04286,0.03886,0.04200,0.04562,0.04688,0.04333,0.04568,0.04611,0.04830,0.04860,0.04941,0.05112,0.04950,0.04688,0.05169,0.05247,0.05599,0.05057,0.05495,0.05455,0.05585,0.05492,0.05489,0.05418,0.05623,0.05948,0.05596,0.06067,0.05725,0.05656,0.06137,0.06230,0.06001,0.06789,0.06750,0.06873,0.06561,0.06878,0.06679,0.06939,0.07094,0.06956,0.06865,0.07507,0.07695,0.07758,0.07947,0.07635,0.07881,0.07817,0.08168,0.08066,0.08750,0.08951,0.08956,0.09403,0.09756,0.09836,0.10364,0.11207,0.11606,0.13136,0.14360,0.17319,0.21420,0.27017,0.34291,0.43074,0.51912,0.59940,0.66895,0.71914,0.75979,0.79461,0.82302,0.84414,0.86432,0.87917,0.88823,0.90490,0.90748,0.90929,0.91120,0.91984,0.92563,0.92033,0.92047,0.92811,0.93138,0.93057,0.92998,0.93393,0.93219,0.93583,0.93588,0.93852,0.93740,0.93998,0.94129,0.94125,0.94296,0.94218,0.94401,0.94555,0.94693,0.94797,0.94827,0.94535,0.94694,0.94635,0.94998,0.95167,0.94514,0.95030,0.95079,0.94888,0.94767,0.95399,0.95439,0.95494,0.95493,0.95140,0.95095,0.95331,0.95148,0.95608,0.95442,0.95307,0.94187,0.93457,0.92416,0.90101,0.87616,0.82514,0.77242,0.70212,0.64445,0.59238,0.54527,0.51944,0.50459,0.50768,0.52760,0.55669,0.61121,0.67446,0.74421,0.80153,0.84270,0.88067,0.90494,0.92629,0.93850,0.95481,0.95641,0.96155,0.96595,0.96374,0.97067,0.96878,0.96962,0.96970,0.96919,0.97645,0.96998,0.96934,0.97360,0.97454,0.97462,0.97598,0.97358,0.97298,0.97817,0.97477,0.97594,0.97211,0.97224,0.97579,0.97560,0.97539,0.98010,0.97653,0.97373,0.97551,0.97673,0.97852,0.98019,0.97952,0.97597,0.97340,0.97704,0.97843,0.97967,0.97844,0.97946,0.98172,0.97744,0.98169,0.98023,0.97937,0.98098,0.98240,0.97941,0.98134,0.98151,0.98459,0.97673,0.96922,0.95919,0.94367,0.91764,0.87786,0.81635,0.75865,0.69718,0.64366,0.60288,0.58055,0.57436,0.59134,0.61957,0.66381,0.72676,0.79101,0.84136,0.88317,0.91702,0.93982,0.96234,0.97088,0.98207,0.98571,0.99144,0.99101,0.99433,0.99289,1.00169,0.99790,0.99723,0.99491,0.99719,0.99440,0.99738,0.99417,0.99356,0.99863,0.99137,0.99877,0.99802,0.99390,0.99984,0.99792,0.99966,0.99734,0.99934,0.99935,0.99857,0.99926,0.99776,1.00068,0.99345,0.99644,0.99816,1.00274,0.99921,0.99627,0.99942,0.99695,1.00130,0.99848,0.99747,0.99926,1.00116,0.99957,1.00356,0.99916,0.99520,1.00139,1.00164,0.99853,0.99484,0.99746,0.99472,0.98657,0.98807,0.97782,0.95381,0.93318,0.88648,0.83269,0.76805,0.70941,0.65783,0.62600,0.60773,0.61239,0.63159,0.66487,0.72092,0.77709,0.83096,0.88081,0.91773,0.93769,0.95865,0.97268,0.98237,0.99506,0.99709,1.00309,0.99923,1.00123,1.00246,1.00098,1.00194,1.00302,0.99844,1.00329,1.00166,1.00201,1.00114,1.00755,1.00013,1.00268,0.99887,1.00206,1.00373,1.00128,1.00114,1.00123,0.99973,1.00594,1.00026,1.01341,1.00985,1.00857,1.01068,1.01152,1.00921,1.00880,1.00978,1.01043,1.00511,1.00809,1.00773,1.00831,1.00763,1.00805,1.00975,1.00611,1.00282,1.00835,1.00893,1.00084,1.00377,1.00202,1.00621,1.00854,1.00528,1.00188,0.99189,0.98923,0.96973,0.94672,0.91004,0.85751,0.80287,0.73356,0.68287,0.64153,0.61424,0.60974,0.61586,0.65428,0.70480,0.75759,0.81830,0.86616,0.90744,0.93563,0.95557,0.97409,0.98687,0.99115,1.00165,1.00515,1.00315,1.00742,1.00508,1.00932,1.00324,1.00791,1.00428,1.00448,1.00638,1.00557,1.00523,1.00559,1.00607,1.00742,1.00528,1.00634,1.00493,1.01065,1.00708,1.00685,1.00454,1.00896,1.01115,1.00487,0.99927,1.00558,1.00426,1.00820,1.00317,1.00500,1.00237,1.00569,1.00533,1.00674,1.00023,1.00318,1.00486,1.00284,1.00393,1.00558,1.00007,1.00476,1.00380,1.00148,1.00189,1.00038,1.00091,0.99960,0.99975,0.99899,0.99079,0.98891,0.98058,0.95696,0.92267,0.88183,0.82630,0.75437,0.69568,0.64693,0.60724,0.59809,0.60178,0.62125,0.66117,0.71724,0.77794,0.83461,0.87944,0.90947,0.93846,0.95763,0.97429,0.98374,0.98292,0.99368,0.99117,0.99319,0.99051,0.99585,0.99567,0.99931,0.99074,0.99205,0.99256,0.99551,0.99751,0.99015,0.99047,0.98997,0.98935,0.99144,0.99298,0.98964,0.99391,0.99220,0.99243,0.98929,0.99169,0.99697,0.99119,0.99398,0.98875,0.99218,0.99095,0.98985,0 [... cut ...] 11,0.00741,0.00587,0.00960,0.00290,0.00432,0.00961,0.00397,0.00666,0.00431,0.01023,0.00535,0.00621,0.00307,0.00442,0.00310,0.00370,0.00420,0.00635,0.00484,0.00152,0.00449,0.00548,0.00759,0.00371,0.00455,0.00505,0.00350,0.00212,0.00584,0.00634,0.00494,0.00635,0.00218,0.00775,0.00440,0.00791,0.00521,0.01009,0.01477,0.01787,0.03098,0.04860,0.06929,0.09178])

prof = SingleProfile(prof_data)
prof.plot()
plt.show()
```

Produces the plot above on master.  This results in a DRGS analysis like:

[image]

randlet, 2021-04-15:
Although I'm confused because the central peak is detected in the right place, but the segment centres are not calculated correctly.

jrkerns, 2021-04-15:
Can you print out the SEGMENT_X_POSITIONS_MM property of the DRGS?

randlet, 2021-04-15:
`SEGMENT_X_POSITIONS_MM = (-60, -40, -20, 0, 20, 40, 60)`

randlet, 2021-04-15:
Ok, I see why the plot is different.   

```
    def plot(self, x: int=50) -> None:
        """Plot the profile."""
        peak_idx, peak_props = find_peaks(self.values, fwxm_height=x/100, max_number=1)
        ...
        # uses peak_idx directly
        plt.vlines(peak_idx, ymin=peak_props['peak_heights'][0]-peak_props['prominences'][0], ymax=peak_props['peak_heights'][0], color="red")
        ...
```

but in fwxm_center it uses the `right_ips` and `left_ips` to calculate the center:

```
    def fwxm_center(self, x: int=50, interpolate: bool=False) -> Tuple[NumberLike, NumberLike]:
        """Return the center of the FWXM.

        See Also
        --------
        fwxm() : Further parameter info
        """
        peak_idx, peak_props = find_peaks(self.values, fwxm_height=x/100, max_number=1)
        fwxm_center_idx = (peak_props['right_ips'][0] - peak_props['left_ips'][0])/2 + peak_props['left_ips'][0]
        if interpolate:
            return fwxm_center_idx, self._values_interp[int(fwxm_center_idx*self.interpolation_factor)]
        else:
            fwxm_center_idx = int(round(fwxm_center_idx))
            return fwxm_center_idx, self.values[fwxm_center_idx]
```
So `plot` uses 512 as the location to draw its line, but fwxm_center calculates a value of 546 which is not correct and that throws off the segment position calculation.